nordicsemi/Android-BLE-Library #106: Can't connect the device on first time

affcue25, 2019-05-09:
I am making a ECG app. And i have a ECG sensor BLE device. When i try to connect it on first time, it held on "Connecting". And when i kill the app from tasks, opens it again and click on connect, then connection established.
Why it doesnt connect on first time?
Is there any problem here:
```
mBleManager.connect(device)
				.useAutoConnect(shouldAutoConnect())
				.retry(3, 100)
				.enqueue();
```

philips77, 2019-05-09:
Hello, do you scan for it before you start connecting?

affcue25, 2019-05-09:
> Hello, do you scan for it before you start connecting?

Yes, when i click "connect" it opens a scan dialog, where it show me the scanned BLE devices that are available. Then i click on my device.

philips77, 2019-05-09:
Then I gave no idea why would it not connect. Does it fail on a timeout 30 sec? Do you have logs from the app that the BLE Library is producing?

affcue25, 2019-05-09:
> Then I gave no idea why would it not connect. Does it fail on a timeout 30 sec? Do you have logs from the app that the BLE Library is producing?

No, its just held on "connecting" until i restart the app. But on second  attempt it always connect the device. But it works fine on some android devices. :/

JamesDougherty, 2019-05-16:
BLE on Android is not the best so without a log it would just be all guesses. In your mBleManager class override the following method and look at the logcat to see what it's outputting:

`@Override
public void log(final int priority, @NonNull final String message) {
    Log.d("BleManager", "[BLE LOG] " + message);
}`

philips77, 2022-03-21:
I'm closing the issue due to inactivity.

closed 2022-03-21 by philips77