nordicsemi/Android-BLE-Library #108: Unable to perform OTA on Nexus 5 (Android 6.0.1)

lookashc, 2019-05-21:
Hi guys,

We’re experiencing issues with our OTA mechanism for our BLE-powered devices, while using Android 6 or earlier (everything works fine on Android 7 and later versions).

The process of OTA goes like this:

- We send command to our OTA_START characteristics in OTA service
- Device reboots in a bootloader mode
- We perform update of device’s firmware
- We ask the device to reboot itself back to the non-bootloader mode
- If any of the above stages fail or hit timeout, we’re making full recovery by disconnecting - from the device, reconnecting and starting the whole process from scratch
- Our device has LED to indicate if its in bootloader or not

The list of issues we’re experiencing goes as follows:

- On devices powered by Android 6 (our reference is Nexus 5 in this case), the very first stage reboot to bootloader - is failing in 100% of cases. First we get status 133 while trying to send OTA start command, and after disconnection we cannot reconnect again (we get Error (0x1): GATT INVALID HANDLE while trying to read any characteristics). See [log](https://drive.google.com/file/d/1vf43fI3VDqLicK243FLhLgvTmv5jK2tf/view) to see the logs.
- We modified the process by adding clean BLE cache command before attempting reconnection. This caused initial stage to pass - device was able to enter bootloader and firmware updated started normally.
- However, it fails on the last step - rebooting after finished firmware update. Our bootloader check returns true, while the LED indicates that the device is not in the bootloader anymore.
- Furthermore we’re getting incorrect response from one of characteristics - payload E0-34-5C-4E-B6-CF-40-B6-89-49-D0-A1-00-00-00-02. This is not only incorrect payload, it looks like it’s a random data returned by BLE stack (analysis of HCI logs revealed that this payload doesn’t come in any frame). See [log](https://drive.google.com/file/d/1AFQWcYM_jPM8OZ2xJKvMNFiAXXi7Vuf7/view?usp=sharing) for details.
- Adding clean BLE cache command before second disconnection causes other issue - connection never gets established as the process hangs before the troubled characteristics is even read. See [log](https://drive.google.com/file/d/1kJq6k50WxkUeU2u8msX4lKzbqjqfFAGR/view?usp=sharing) for details. 
- In addition, what we checked is that cleaning the BLE cache causes troubles on newer Android phones (for ex. OnePlus 3T with Android 7 started to fail on initial bootloader reboot).

Do you think the clean BLE cache is the correct approach to fix such issues on Android versions pre-7.0? If so - do you have any hints why the process hangs after second clean? And if you think this is incorrect - how can we approach the problem in a different way?

philips77, 2019-06-18:
Hi, Sorry it takes so long. I remember about your issue, but I'm very busy with other tasks. Did you do any progress on this?

lookashc, 2019-06-18:
Hi @philips77 unfortunately not, for the time being we had to disable OTA for Android 5 & 6. We can go back to the issue this week, if you have any hints that would help us, it will be great :)

closed 2022-02-11 by lookashc

philips77, 2022-02-14:
Finally closed! ;)