nordicsemi/Android-BLE-Library #109: Unable to read correct int values from Data in onDataReceived callback

jimmy-jose, 2019-05-27:
I am new to BLE and I am using [BLE Peripheral Simulator](https://play.google.com/store/apps/details?id=io.github.webbluetoothcg.bletestperipheral&hl=en) to test out BLE-Library. With the help of Blinky example, I was able to establish the connection and receive a call back from the simulator. But the issue I am facing is that I am not able to get the correct values that are being notified from the simulator. I am trying to get [HeartRateMeasurement](https://github.com/WebBluetoothCG/ble-test-peripheral-android/blob/master/app/src/main/java/io/github/webbluetoothcg/bletestperipheral/HeartRateServiceFragment.java)

The simulator is setting the value as shown [below](https://github.com/WebBluetoothCG/ble-test-peripheral-android/blob/d6477927e7fe90bb03c18632d52e5846a13bf387/app/src/main/java/io/github/webbluetoothcg/bletestperipheral/HeartRateServiceFragment.java#L101) in the simulator

`private static final int HEART_RATE_MEASUREMENT_VALUE_FORMAT = BluetoothGattCharacteristic.FORMAT_UINT8;
mHeartRateMeasurementCharacteristic.setValue(newHeartRateMeasurementValue,
              HEART_RATE_MEASUREMENT_VALUE_FORMAT,
/* offset */ 1);`

And I am trying to read it in my callback as shown below
`public void onDataReceived(@NonNull BluetoothDevice device, @NonNull Data data) {
        Log.d(TAG, "onDataReceived: "+ data.getIntValue(Data.FORMAT_UINT8,1));
    }`

But this is always giving me 60. I think I am doing something wrong here. Please tell me what is the right way to read int values from Data.

Please guide me in the right direction.

Here is my [code](https://github.com/jimmy-jose/SHBLETest).

philips77, 2019-05-27:
Hi, do you set the correct value in the BLE Simulator? I see that value 60 is used as default on one of the screenshots in Google play.
Btw, we recommend using nRF Connect for such purposes. It also allows you to configure a hat server and start advertising.

jimmy-jose, 2019-05-28:
Hi @philips77 Thanks for looking into this. The issue was with the BLE Simulator. Sorry for wasting your time.

closed 2019-05-28 by jimmy-jose