nordicsemi/Android-BLE-Library #110: Reason for disconnect is unknown

sssemil, 2019-06-03:
It's impossible to know the reason for device disconnect. Would it be possible to implement some sort of internal mapping of the Android's GATT errors/states, that is passed along with the 'onDeviceDisconnected' call? Here are some examples of reasons: user disconnected on the remote device, user disconnected on the local device, connection timed out, etc. Or generally speaking it would be nice be able to distinguish intentional and unintentional disconnects.

Related https://github.com/NordicSemiconductor/Android-BLE-Library/pull/21.

pblitz, 2019-06-07:
maybe we can actually support and do a PR for this. Any preference on how to do this? Adding a parameter to onDeviceDisconnected will break the API. Is that okay? 
Or should we add a onDeviceDisconnectedWithReason() (that looks horrible tho :/ )

philips77, 2019-06-07:
I guess I would add another Reason in FailCallback for disconnection initiated by remote device leaving the current status -1 for locally initiated. Timeout already has it's reason. This also includes adding a new Exception for synchronous calls. Indeed send like a breaking change, so would increase version to at least 2.2, but worth doing.

commit 6f652f93 (2020-03-31): Merge pull request #149 from workaroundgmbh/disconnect_status

Add extended disconnect callback (closes #110)
closed 2020-03-31 by philips77