nordicsemi/Android-BLE-Library #111: Question about timeout on missing indication after write in control point

cyberlussi, 2019-06-05:
Hi,
I am wondering what would be a generic and compact way to monitor timeouts of missing indication after writing a control point characteristic. I looked through all the examples I am aware of (nRF Blinky, nRF Toolbox) but I have not found anything. 
I assume the Android-BLE-Library itself is not aware of the definition of control points (which are write + indicate) as this is more in the scope of higher layers and I know that depending on the service the behavior of a control point my be slightly different (e.g. RACP), but as I am new to Android and Java I am just wondering how to implement an elegant way to start a timeout monitor with configurable limit when writing to the control point and how to cancel the monitor when the (expected) indication arrives. 
Do you have a best practice example for this?

philips77, 2019-06-06:
Hello,
Timeouts are supported for notifications and indications in this library. Check out waitForIndication(...).timeout(..).enqueue() in BleManager.

philips77, 2019-06-06:
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/5e0e2f08c309a6de2376d9b8705c83f9e9a80d56/ble/src/main/java/no/nordicsemi/android/ble/BleManager.java#L1156-L1158
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/5e0e2f08c309a6de2376d9b8705c83f9e9a80d56/ble/src/main/java/no/nordicsemi/android/ble/WaitForValueChangedRequest.java#L77-L83

cyberlussi, 2019-06-06:
Oh, okay. Got it. Works like a charm. Easier than I thought! Thank you. 

Note:
It would be great to have some wiki pages explaining the concepts and typical use cases for this great library. I mean, I am reading and working through the example projects which are referenced. But it would make some things easier to understand having those topics expained in natural language. Or do I missed some documentation e.g. in Nordic's DevZone?

closed 2019-06-06 by cyberlussi