nordicsemi/Android-BLE-Library #113: Auto connect not working in foreground service when link loss while screen off

rickymohk, 2019-06-26:
I have a Activity that starts a foreground service and the foreground service connect to BLE devices with `useAutoConnect(true)`. 

If I stop the activity by clicking the home button, turn off the screen and cut the bluetooth link by taking the bluetooth device far away from the phone, the  bluetooth device won't auto reconnect until I turn on the screen.  Once I turn on the screen, the device auto reconnects without needing to unlock the screen and launch the activity. 

If I do not stop the activity but just turn the screen off, the bluetooth device can auto reconnect without needing to turn the screen back on.

All the BLEManager instance are created in the foreground service and connect device in the foreground service. The activity does not handle any of those.

philips77, 2019-06-26:
Interesting issue. But hard to say why. Did you try on other phones, perhaps with older Android version? It may be a Dose thing.
I wouldn't say that running activity should have any impact on foreground service...

rickymohk, 2019-06-26:
Tried with Lenovo A7000 plus Android 6.0. No issue, reconnect normally. I also discovered on the CAT S41 that when there were 2 connected BLE devices and only one link loss, it can auto reconnect without problem, just the range become a little bit shorter. I step through the debugger when the problem occurs and found that it successfully ran through `gatt.connect()`, but the BLE device did not get connected until screen on.

shiehnpin, 2019-06-26:
While screen is off, new scans are allowed to be
created but scan instances will not receive results
until the screen is turned on.

Stack overflow reference: https://stackoverflow.com/questions/48077690/ble-scan-is-not-working-when-screen-is-off-on-android-8-1-0

rickymohk, 2019-06-26:
> While screen is off, new scans are allowed to be
> created but scan instances will not receive results
> until the screen is turned on.
> 
> Stack overflow reference: https://stackoverflow.com/questions/48077690/ble-scan-is-not-working-when-screen-is-off-on-android-8-1-0

This should not be the case since my app is not scanning, just reconnecting to a known device. It does successfully reconnect with screen off in some cases (e.g. the parent activity is running/ on Android 6.0/ another BLE device is connected).

rickymohk, 2019-06-26:
Turns out if I start a scanner once all bluetooth devices are link lost, the problem is gone. I guess the system somehow put the bluetooth adapter into some kind of "sleep" state if there are no more bluetooth actions occurring. If I start a scanner to keep it "awake", it can auto connect normally.

closed 2020-03-11 by philips77