nordicsemi/Android-BLE-Library #116: MTU Transmission issue

Galaxy-Man, 2019-07-15:
Hi

When using an ESP32 the MTU negotiation is ok, it starts with the default 23 bytes then is switched to 500 by the Android application but the application (or Android) keeps sending only 20 bytes.
If the service is restarted it works again. Could it be a bug within the Nordic BLE library.

Any help would be appreciated.

philips77, 2019-07-16:
What do you mean by

> If the service is restarted it works again.

Galaxy-Man, 2019-07-16:
Hi philips77

When the Service using the BLE Library is restarted. The other developers may have found a solution and trying an update to the Applciation code around the BLE library.

This is the link in github for the project.

[https://github.com/mitchwongho/ESP-Alerts-for-Android/issues/3](url)

philips77, 2019-10-22:
Hi, did you manage to solve this issue? Sorry for the late response, but I was busy with another projects.

Galaxy-Man, 2019-10-22:
Hi I believe it has been resolved for now. The only issue is the Andorid reconnecting, we think that is in the Android Java code rather than the library.

Thanks

Ianandra

closed 2019-10-22 by philips77

Galaxy-Man, 2019-10-22:
Hi quick question how to reset MTU memory when the connection is lost, but i don't know how to fix that and haven't looked into that very much, i don't program in Java that much and almost never for Android, but since this seems to point out the underlying problem, i guess it should be easy enough to fix for someone with more knowledge about the used libraries.

philips77, 2019-10-22:
What do you mean be reset MTU memory?