nordicsemi/Android-BLE-Library #117: WriteCharacteristic not performed because thread problem with TaskQueue.

balazs-muller, 2019-08-01:
We connect with Ebike HMIs and the connections freeze about 10-45 minute because one writeCharacteristic method not performed.
I think the problem is in the BleManager.java mTaskQueue object. This is a LinkedList<>(), so not thread-safe.

So after a while communication bug occurs.
I think: While enqueue calls and perform an add action to mTaskQueue object, on another thread mTaskQueue.remove() calls (BleManager.java line:3169). So remove fails and NoSuchElementException caught and the WriteCharacteristic not continues.

I found a simple solution to fix this issue.
Set the mTaskQueue type LinkedList<>() to LinkedBlockingDeque<>();

In my case issue fixed.

Thanks a lot.

balazs-muller, 2019-08-05:
Hi,
Expected to be fixed for this task? Or you need more information about the problem?

Thanks.

philips77, 2019-08-05:
Hi,
I understand the problem and the solution seems OK. I'm actually busy with other tasks, that's why I haven't release a new version yet. Sorry for that.

balazs-muller, 2019-08-12:
Hi,
Thanks!

Could you estimate the next release date?

philips77, 2019-08-12:
No, sorry. I hope in August.

balazs-muller, 2019-08-30:
Hi,

You may now know the release date? Our customers are looking forward to it.

Thanks.

balazs-muller, 2019-09-27:
Hi,

In the near future, we are planning to release the new version of our application that would rely on this fix.

Do you think that there is a chance that the problem will soon be solved on your side, or should we try and find a workaround for now?

Thank you and best regards

philips77, 2019-09-27:
Hi, sorry that i made you wait that long. I'm finishing nRF Mesh library for iOS at the moment, after which I'll fix all other remaining issues in other libraries and apps. I'll start with this library. Please, expect a release in the coming week, max 2.
Btw, we've find another issue, when a characteristic is written without response to transfer large amount of data. The library switches to main thread for each request which makes the upload slow (1-2 packets per connection interval). This week also be addressed in the next release.

philips77, 2019-10-22:
Hi, sorry, I was out of office for those 2 weeks I promised to release the app :(

balazs-muller, 2019-10-22:
Thank you for lettings us know. We created an interim implementation to bypass the problem for our current release, however, it will be great to use your solution when it is released.

philips77, 2019-11-25:
Fixed internally.

closed 2019-11-25 by philips77