nordicsemi/Android-BLE-Library #119: Severe packet loss on Android Pie

mcolunas, 2019-08-12:
Hello,

 We are facing an issue by having a lot of packet loss (UART nrf52832) on two sensors connected to an Android Application (Android Pie 9.0). The tablet model that runs the application is a Galaxy Tab A (2019) Samsung T-515 (already supports BLE 5.0). Moreover that behavior does not happen on other tablet/smartphones models that supports BLE 4.2 such as the previous version Galaxy Tab A (2019) Samsung T-585.

philips77, 2019-08-12:
Please, create an issue on DevZone. I think lowering the Packet Receipt Notifications to 1 or few would help, but that may be an issue on bootloader side. The support team would be more helpful.

philips77, 2019-10-22:
Hello,
Sorry for the late reply. This looks like a threading issue. The newer devices are able to send/receive data faster and juggling with threads and handlers seems to be causing an issue here. Could you provide a test app and fw on nRF52832 where we could test it and apply a fix to the library?

waqarul, 2021-05-18:
@philips77 
Do we have any solution for this? 

We have faced similar issue in which packet was not received on remote device and even Bluetooth Analyzer "ElliSys" was not able to detect. Though I can see from the logs that data was sent. We are using _WRITE_TYPE_NO_RESPONSE_ on data characteristics and have different characteristics to Read & Write data.

philips77, 2021-05-20:
What phone are you using? Also Android 11? Did you try a different phone? Can you share logs?

philips77, 2022-03-21:
I'm closing the issue due to inactivity.

closed 2022-03-21 by philips77