nordicsemi/Android-BLE-Library #120: initialize called twice

hearsilent, 2019-08-20:
BleManagerGattCallback `initialize` called twice when I connect ble device.

**Test Device 1**
1. Device Name: Pixel 3 XL
2. OS: Android Q beta 6

**Test Device 2**
1. Device Name: Pixel 2
2. OS: Android P

Library Version: 2.1.1

philips77, 2019-12-06:
Threr's only 1 place where this method is called, it's after service discovery is complete and the required services were found. Perhaps you connected to the device twice in a row?

rorold, 2020-09-16:
I'm seeing this same behaviour with Library version 2.2.4 - I came to the same conclusion that I must be connecting twice somehow as well.

But I can see in the logs from my connection observer, onDeviceConnecting and onDeviceConnected are both called once, but isRequiredServiceSupported is called twice

rorold, 2020-09-16:
Incidentally this is with a device that requires bonding.

Looking in BleManagerHandler, I noticed that in the bonding broadcast receiver, you're checking if (!servicesDiscovered && !serviceDiscoveryRequested)  and then setting serviceDiscoveryRequested = true before posting the discoverServices task, whereas in onConnectionStateChange you're only checking if (!serviceDiscoveryRequested) and then setting serviceDiscoveryRequested = true; inside the discoverServices task, which is posted with a delay. I suspect this is the cause.

philips77, 2020-09-16:
Thank you for pointing it. I will look into that when I have time.

rorold, 2020-09-16:
Thanks! Overriding getServiceDiscoveryDelay to return 0 helps with the timing in my case, but it still feels like the logic in BleManagerHandler could be tightened up a bit.

zetokz, 2021-01-20:
I believe it is because `serviceDiscovery` callback call is called on some devices several seconds after connection or pairing. We have such an issue on Pixel 2, each time we connect to our paired Bluetooth device using this Pixel 2 phone it leads to service discovery callback call and `initialize` call in a result.

philips77, 2021-01-20:
@zetokz you're saying it is related to/caused by #273?

zetokz, 2021-01-20:
> @zetokz you're saying it is related to/caused by #273?

@philips77 yep, that is my assumption
If we would have logs here we would see if the `onServicesDiscovered` callback is called, if so then #273 is the root cause

closed 2022-03-21 by philips77