nordicsemi/Android-BLE-Library #127: Android-BLE-Library not working properly

dominicbar, 2019-09-11:
after updating ble library 2.1.1, mesh services are not getting detected(giving null), so after provisioning phone thinks the bluetooth device is still unprovsioned,
Note that this issue happens for xiaomi redmi note 3 device

In nrf mesh we upgraded the android ble library to 2.1.1, we could reproduce the issue there also
[image]

roshanrajaratnam, 2019-09-11:
@dominicbar Since you have upgraded the the library, I suppose you need to refresh the Gatt cache, if not Android will use cached services from the first connection.

dominicbar, 2019-09-11:
We did the above, most of the phones its working, but 2 xiaomi redmi note 3 are giving the same problem

dominicbar, 2019-09-11:
but ya once we downgrade the ble library, the redmi note 3 phone works fine

roshanrajaratnam, 2019-09-11:
What you may also do is set up retry events, so the device will try to connect multiple times which is supported on the latest BLE Library

philips77, 2019-10-22:
Hi, could you post here logs from the BLE Library 2.x during the connection? Override the `log` method in the `BleManager`. Let's see what is going on there.

philips77, 2019-11-27:
Hi, we have migrated nRF Mesh to Android BLE Library 2.1.1 and fixed all the issues. Please, take a look at them. All should be working now.

closed 2019-11-27 by philips77