nordicsemi/Android-BLE-Library #129: Public key response mismatch when switching no.nordicsemi.android:ble dependency version

WuLFGit, 2019-09-24:
---

I tried to provision a device multiple times failing using no.nordicsemi.android:ble:2.1.1 version. Switching back to no.nordicsemi.android:ble:1.2.0 makes provisioning work back again. What i noticed is a difference between response byte array i receive when sending public key to my device. 

May i ask you why this behaviour happens? It sounds weird to me that switching ble library version result in different byte array response received from the device.

**Logs / Screenshots**
Here some screenshot about responses i receive when public key is sent:

Public key sent response on no.nordicsemi.android:ble:2.1.1. As you can notice, on position 1 i expect value 3 but i receive -99:
[image]

Public key sent response on no.nordicsemi.android:ble:1.2.0. As you can notice, on position 1 there is the expected value 3:
[image]

Responses length is different also.

Thank you in advance for you support.

roshanrajaratnam, 2019-09-25:
Duplicate of https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/issues/250

philips77, 2019-10-30:
I will take the issue in this project.

philips77, 2019-10-30:
Could you post the logs from your app (that's using BLE library version 2.x) and some code snippets how do you request and set the callbacks? Do you set MTU to 250+? Are you sure the response you get using version 2 isn't a correct response, but the request is invalid, so the device sends an error message instead of Composition Data Status?

philips77, 2019-10-30:
I hoped that you have this integration with BLE library 2.x somewhere on another branch. As I said, to help you I need the logs and snippet of your code to check if your implemention is correct.
As I understand, the issue is that the old version of the library returns correct value of the received notification (66 bytes, not 20, as would be received issuing mtu=23), but the new version returns only 28 bytes. Isn't that right? The values are then combined from multiple notifications using the mesh proxy protocol, but as you're reporting this issue for the BLE library, I understand that the very low layer has an issue, not the concatenation. But now, when you're saying that the mtu is set to 23, that would mean that the status displayed above are already combined from 2+ notifications. As you see, without the logs I can't really say much.

philips77, 2019-11-08:
I've migrated nRF Mesh for Android to use BLE Library 2.1.1 and could not replicate this issue.

philips77, 2019-11-25:
Issue fixed in nRF Mesh library. It was a MTU issue. The `mMtu` variable wasn't reset on disconnection, so next time the same manager was used for a different connection when the MTU was successfully set on the previous time, the MTU request was no longer performed.

philips77, 2019-11-25:
Fixed internally.

closed 2019-11-25 by philips77