nordicsemi/Android-BLE-Library #130: SDK 22 onConnectionStateChange after connected immediately (40ms) called with disconnected

KORuL, 2019-10-10:
I am trying to connect to a BLE device. Everything goes well on android sdk> 23 and sdk <20, however I noticed that an error occurs on MEIZU M1 Note with Android 5.1 (sdk 22). The first time you connect a BLE device, everything goes well, the device connects, Pairing occurs, the services are located, and then I use the device. BUT! if you disconnect the device and try to connect it again, the `onConnectionStateChange` function with the `Connected` parameter is called and is immediately called with `disconnected` status.
I reviewed all issues and stackoverfow, I have already tried all the methods described, but I still can’t find a solution. Tell me out of the situation, please.

philips77, 2019-10-11:
This looks like an issue on Meizu phone... I guess something with reinitiating bonding. Did you try the same with a device without bonding?
If you're able to collect traces, could you report this on Nordic's DevZone specifying which Nordic SDK are you using in nRF5 devices? I think we may have this phone to test it at our office.

KORuL, 2019-10-12:
Logcat from BleManager

```D/BluetoothDevice: mAddress: 0C:61:CF:C9:C8:E8
    mAddress: 0C:61:CF:C9:C8:E8
    mAddress: 0C:61:CF:C9:C8:E8
D/BluetoothAdapter: isEnabled
V/BleManager: Connecting...
D/BleManager: gatt = device.connectGatt(autoConnect = false)
D/BluetoothDevice: mAddress: 0C:61:CF:C9:C8:E8
D/BluetoothGatt: connect() - device: 0C:61:CF:C9:C8:E8, auto: false
    registerApp()
D/BluetoothGatt: registerApp() - UUID=c23800c7-32f9-4612-9aa7-5fbcd9738fb1
D/BluetoothGatt: onClientRegistered() - status=0 clientIf=1
D/BluetoothDevice: mAddress: 0C:61:CF:C9:C8:E8
D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=1 device=0C:61:CF:C9:C8:E8
D/BluetoothDevice: mAddress: 0C:61:CF:C9:C8:E8
D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=1 device=0C:61:CF:C9:C8:E8
D/BleManager: [Callback] Connection state changed with status: 0 and new state: 2 (CONNECTED)
D/BluetoothDevice: mAddress: 0C:61:CF:C9:C8:E8
D/BluetoothDevice: mAddress: 0C:61:CF:C9:C8:E8
I/BleManager: Connected to 0C:61:CF:C9:C8:E8
D/BluetoothDevice: getName: name = Contour7804-2203787
D/BluetoothDevice: getName: name = Contour7804-2203787
D/RemoteViews: mActions is repeat
    mActions is repeat
D/RemoteViews: mActions is repeat
    mActions is repeat
D/RemoteViews: mActions is repeat
    mActions is repeat
    mActions is repeat
D/RemoteViews: mActions is repeat
V/SettingsInterface:  from settings cache , name = notification_fixed_font_color , value = null
D/Notification: fixedFontColor clear=true
D/BluetoothDevice: getBondState: state = 12
D/BleManager: wait(1600)
I/art: Background sticky concurrent mark sweep GC freed 71578(3MB) AllocSpace objects, 3(172KB) LOS objects, 2% free, 69MB/71MB, paused 9.509ms total 66.775ms
D/BleManager: [Callback] Connection state changed with status: 0 and new state: 0 (DISCONNECTED)
I/BleManager: Disconnected
D/BleManager: gatt.close()
D/BluetoothGatt: close()
    unregisterApp() - mClientIf=1
D/BluetoothDevice: getName: name = Contour7804-2203787
D/BluetoothDevice: getName: name = Contour7804-2203787
D/RemoteViews: mActions is repeat
    mActions is repeat
D/RemoteViews: mActions is repeat
    mActions is repeat
D/RemoteViews: mActions is repeat
D/RemoteViews: mActions is repeat
    mActions is repeat
D/RemoteViews: mActions is repeat
V/SettingsInterface:  from settings cache , name = notification_fixed_font_color , value = null
D/Notification: fixedFontColor clear=true```

KORuL, 2019-10-12:
This is the first connection with a non-pairing device, first bound and then everything works fine, until the first disconnection of the device, the next time it fails

```
D/BluetoothDevice: mAddress: 0C:61:CF:C9:C8:E8
    mAddress: 0C:61:CF:C9:C8:E8
D/BluetoothDevice: mAddress: 0C:61:CF:C9:C8:E8
    mAddress: 0C:61:CF:C9:C8:E8
D/BluetoothDevice: mAddress: 0C:61:CF:C9:C8:E8
D/BluetoothDevice: mAddress: 0C:61:CF:C9:C8:E8
D/BluetoothDevice: mAddress: 0C:61:CF:C9:C8:E8
    mAddress: 0C:61:CF:C9:C8:E8
D/BluetoothDevice: mAddress: 0C:61:CF:C9:C8:E8
    mAddress: 0C:61:CF:C9:C8:E8
D/BluetoothDevice: mAddress: 0C:61:CF:C9:C8:E8
    mAddress: 0C:61:CF:C9:C8:E8
D/BluetoothDevice: mAddress: 0C:61:CF:C9:C8:E8
    mAddress: 0C:61:CF:C9:C8:E8
D/BluetoothDevice: mAddress: 0C:61:CF:C9:C8:E8
    mAddress: 0C:61:CF:C9:C8:E8
D/BluetoothDevice: mAddress: 0C:61:CF:C9:C8:E8
    mAddress: 0C:61:CF:C9:C8:E8
D/BluetoothDevice: mAddress: 0C:61:CF:C9:C8:E8
    mAddress: 0C:61:CF:C9:C8:E8
D/BluetoothDevice: mAddress: 0C:61:CF:C9:C8:E8
    mAddress: 0C:61:CF:C9:C8:E8
D/BluetoothDevice: mAddress: 0C:61:CF:C9:C8:E8
    mAddress: 0C:61:CF:C9:C8:E8
D/BluetoothDevice: mAddress: 0C:61:CF:C9:C8:E8
    mAddress: 0C:61:CF:C9:C8:E8
D/BluetoothDevice: mAddress: 0C:61:CF:C9:C8:E8
    mAddress: 0C:61:CF:C9:C8:E8
D/BluetoothDevice: mAddress: 0C:61:CF:C9:C8:E8
    mAddress: 0C:61:CF:C9:C8:E8
D/BluetoothDevice: mAddress: 0C:61:CF:C9:C8:E8
D/BluetoothDevice: mAddress: 0C:61:CF:C9:C8:E8
D/BluetoothDevice: mAddress: 0C:61:CF:C9:C8:E8
    mAddress: 0C:61:CF:C9:C8:E8
D/BluetoothDevice: mAddress: 0C:61:CF:C9:C8:E8
    mAddress: 0C:61:CF:C9:C8:E8
D/BluetoothDevice: mAddress: 0C:61:CF:C9:C8:E8
    mAddress: 0C:61:CF:C9:C8:E8
D/BluetoothDevice: mAddress: 0C:61:CF:C9:C8:E8
    mAddress: 0C:61:CF:C9:C8:E8
D/BluetoothDevice: mAddress: 0C:61:CF:C9:C8:E8
    mAddress: 0C:61:CF:C9:C8:E8
D/BluetoothDevice: mAddress: 0C:61:CF:C9:C8:E8
    mAddress: 0C:61:CF:C9:C8:E8
    mAddress: 0C:61:CF:C9:C8:E8
D/BluetoothAdapter: isEnabled
V/BleManager: Connecting...
D/BleManager: gatt = device.connectGatt(autoConnect = false)
D/BluetoothDevice: mAddress: 0C:61:CF:C9:C8:E8
D/BluetoothGatt: connect() - device: 0C:61:CF:C9:C8:E8, auto: false
    registerApp()
    registerApp() - UUID=3ea3092f-28f3-4324-842e-6930c83dc758
D/BluetoothDevice: mAddress: 0C:61:CF:C9:C8:E8
    mAddress: 0C:61:CF:C9:C8:E8
D/BluetoothGatt: onClientRegistered() - status=0 clientIf=2
D/BluetoothDevice: mAddress: 0C:61:CF:C9:C8:E8
D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=2 device=0C:61:CF:C9:C8:E8
D/BluetoothDevice: mAddress: 0C:61:CF:C9:C8:E8
D/BluetoothDevice: mAddress: 0C:61:CF:C9:C8:E8
    mAddress: 0C:61:CF:C9:C8:E8
D/BleManager: [Broadcast] Action received: android.bluetooth.device.action.BOND_STATE_CHANGED, bond state changed to: BOND_BONDING (11)
D/BluetoothDevice: getName: name = Contour7804-2203787
D/BluetoothDevice: getName: name = Contour7804-2203787
D/RemoteViews: mActions is repeat
    mActions is repeat
D/RemoteViews: mActions is repeat
    mActions is repeat
[... 670 lines cut ...]
    unregister, handle(0x7f8f488ca0) (w:715 h:105 s:720 f:0x1 u:0x000b00)
D/OpenGLRenderer: Flushing caches (mode 0)
V/Toast: SHOW: android.widget.Toast$TN@291a6077
V/Toast: HANDLE SHOW: android.widget.Toast$TN@291a6077 mView=null mNextView=android.widget.LinearLayout{33e6a111 V.E..... ......I. 0,0-0,0}
    HANDLE HIDE: android.widget.Toast$TN@291a6077 mView=null
    ADD! android.widget.LinearLayout{33e6a111 V.E..... ......I. 0,0-0,0} in android.widget.Toast$TN@291a6077
D/ViewRootImpl: hardware acceleration is enabled, this = ViewRoot{3b13e526 Toast,ident = 7}
D/Surface: Surface::connect(this=0x7f8f4ab800,api=1)
D/mali_winsys: new_window_surface returns 0x3000
D/Surface: Surface::allocateBuffers(this=0x7f8f4ab800)
D/Surface: Surface::setBuffersDimensions(this=0x7f8f4ab800,w=677,h=105)
D/GraphicBuffer: register, handle(0x7f8f488ca0) (w:677 h:105 s:688 f:0x1 u:0x000b00)
D/Surface: Surface::setBuffersDimensions(this=0x7f8f4ab800,w=677,h=105)
D/GraphicBuffer: register, handle(0x7f8f153b60) (w:677 h:105 s:688 f:0x1 u:0x000b00)
V/Toast: HIDE: android.widget.Toast$TN@291a6077
V/Toast: HANDLE HIDE: android.widget.Toast$TN@291a6077 mView=android.widget.LinearLayout{33e6a111 V.E..... ........ 0,0-677,105}
V/Toast: REMOVE! android.widget.LinearLayout{33e6a111 V.E..... ........ 0,0-677,105} in android.widget.Toast$TN@291a6077
D/OpenGLRenderer: Flushing caches (mode 0)
D/Surface: Surface::disconnect(this=0x7f8f4ab800,api=1)
D/GraphicBuffer: unregister, handle(0x7f8f488ca0) (w:677 h:105 s:688 f:0x1 u:0x000b00)
D/GraphicBuffer: unregister, handle(0x7f8f153b60) (w:677 h:105 s:688 f:0x1 u:0x000b00)
D/OpenGLRenderer: Flushing caches (mode 0)
D/Surface: Surface::setBuffersDimensions(this=0x7fb0a19a00,w=1080,h=1920)
I/art: 
```

KORuL, 2019-10-12:
second connection - false

```
D/BluetoothDevice: mAddress: 0C:61:CF:C9:C8:E8
D/BluetoothDevice: mAddress: 0C:61:CF:C9:C8:E8
D/BluetoothDevice: mAddress: 0C:61:CF:C9:C8:E8
D/BluetoothAdapter: isEnabled
V/BleManager: Connecting...
D/BleManager: gatt = device.connectGatt(autoConnect = false)
D/BluetoothDevice: mAddress: 0C:61:CF:C9:C8:E8
D/BluetoothGatt: connect() - device: 0C:61:CF:C9:C8:E8, auto: false
    registerApp()
    registerApp() - UUID=c220e13a-cbb6-4903-bca2-a202a0e19936
D/BluetoothGatt: onClientRegistered() - status=0 clientIf=2
D/BluetoothDevice: mAddress: 0C:61:CF:C9:C8:E8
D/BluetoothDevice: mAddress: 0C:61:CF:C9:C8:E8
    mAddress: 0C:61:CF:C9:C8:E8
D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=2 device=0C:61:CF:C9:C8:E8
D/BluetoothDevice: mAddress: 0C:61:CF:C9:C8:E8
D/BluetoothDevice: mAddress: 0C:61:CF:C9:C8:E8
    mAddress: 0C:61:CF:C9:C8:E8
D/BluetoothDevice: mAddress: 0C:61:CF:C9:C8:E8
    mAddress: 0C:61:CF:C9:C8:E8
D/BleManager: [Callback] Connection state changed with status: 0 and new state: 2 (CONNECTED)
D/BluetoothDevice: mAddress: 0C:61:CF:C9:C8:E8
I/BleManager: Connected to 0C:61:CF:C9:C8:E8
D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=2 device=0C:61:CF:C9:C8:E8
D/BluetoothDevice: mAddress: 0C:61:CF:C9:C8:E8
D/BluetoothDevice: getName: name = Contour7804-2203787
D/BluetoothDevice: getName: name = Contour7804-2203787
D/RemoteViews: mActions is repeat
    mActions is repeat
D/RemoteViews: mActions is repeat
    mActions is repeat
D/RemoteViews: mActions is repeat
    mActions is repeat
D/RemoteViews: mActions is repeat
D/RemoteViews: mActions is repeat
V/SettingsInterface:  from settings cache , name = notification_fixed_font_color , value = null
D/Notification: fixedFontColor clear=true
D/BluetoothDevice: getBondState: state = 12
D/BleManager: wait(1600)
D/BluetoothDevice: getBondState: state = 12
V/BleManager: Discovering services...
D/BleManager: gatt.discoverServices()
D/BluetoothDevice: mAddress: 0C:61:CF:C9:C8:E8
D/BluetoothGatt: discoverServices() - device: 0C:61:CF:C9:C8:E8
D/BluetoothDevice: mAddress: 0C:61:CF:C9:C8:E8
D/BluetoothDevice: mAddress: 0C:61:CF:C9:C8:E8
    mAddress: 0C:61:CF:C9:C8:E8
D/BleManager: [Callback] Connection state changed with status: 0 and new state: 0 (DISCONNECTED)
I/BleManager: Disconnected
D/BleManager: gatt.close()
D/BluetoothGatt: close()
    unregisterApp() - mClientIf=2
D/BluetoothDevice: getName: name = Contour7804-2203787
D/BluetoothDevice: getName: name = Contour7804-2203787
D/RemoteViews: mActions is repeat
    mActions is repeat
D/RemoteViews: mActions is repeat
    mActions is repeat
D/RemoteViews: mActions is repeat
D/RemoteViews: mActions is repeat
    mActions is repeat
D/RemoteViews: mActions is repeat
V/SettingsInterface:  from settings cache , name = notification_fixed_font_color , value = null
D/Notification: fixedFontColor clear=true
```

KORuL, 2019-10-13:
I find solution!!

If I set shouldClearCacheWhenDisconnected() to true - it help me!
If I set this flag to true, that’s all. The first connection fails, but through the timer the second connection succeeds!

closed 2019-10-13 by KORuL