nordicsemi/Android-BLE-Library #131: ERROR_AUTH_ERROR_WHILE_BONDED ("Phone has lost bonding information")

eliotstock, 2019-10-20:
Hi. I get this warning level logging from the library when I try to read the standard TX power level characteristic from my peripheral. It's coming from `BleManager.onCharacteristicReadSafe()`, here:

```
if (gatt.getDevice().getBondState() != BluetoothDevice.BOND_NONE) {
    // This should never happen but it used to: http://stackoverflow.com/a/20093695/2115352
    Log.w(TAG, ERROR_AUTH_ERROR_WHILE_BONDED);
    mCallbacks.onError(gatt.getDevice(), ERROR_AUTH_ERROR_WHILE_BONDED, status);
}
```

This implies I'm bonded at this point, and indeed my `onBonded()` callback has already been called, but still getting `BluetoothGatt.GATT_INSUFFICIENT_AUTHENTICATION` nonetheless. And your comment here implies this used to be expected (in the bad old days of the Android BLE stack?) but no longer is.

I never get the lambda I pass to `readCharacteristic().with()` called either:

```
// Read the TX power level, so that we can later figure out the path loss from the RSSI.
readCharacteristic(standardTxPowerLevel).with((@NonNull BluetoothDevice device, @NonNull Data data) -> {
    txPowerLevel = data.getIntValue(Data.FORMAT_SINT8, 0);

    Logg.d(TAG, "TX power level: " + txPowerLevel);
}).enqueue();
```

The firmware on the peripheral is mine. I don't think anything's changed in the way I add services. Here's how I init the TX power service:

```
ble_tps_init_t tps_init;
memset(&tps_init, 0, sizeof(tps_init));

tps_init.initial_tx_power_level = TX_POWER_LEVEL;
tps_init.tpl_rd_sec = SEC_MITM;

err_code = ble_tps_init(&m_tps, &tps_init);
APP_ERROR_CHECK(err_code);
```

I would ignore the warning, but I also get the `onError()` callback called a bit later and I do kind of need the characteristic value. Any ideas?

philips77, 2019-10-20:
Are you bonding using "just works" or using and pass code? You sec is SEC_MITM and perhaps this node is insufficient? I'm not an expert in embedded, could you ask this on DevZone?
I think this error should be eliminated, not ignored by the library... But if it didn't work, feel free to do a PR with required fix.

eliotstock, 2019-10-21:
Yep, "just works" bonding. I've been able to read this characteristic in the past. I agree the error shouldn't be ignored by the library. I'll try in the DevZone and see if something's up on the peripheral side.

eliotstock, 2019-10-22:
I've asked here:

https://devzone.nordicsemi.com/f/nordic-q-a/53497/can-t-read-secure-characteristic-after-bonding-just-works-sec_mitm

philips77, 2019-10-22:
Hi, did you try setting SEC_JUST_WORKS instead of SEC_MITM?
https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.sdk5.v15.3.0%2Fgroup__ble__sdk__srv__common.html&cp=5_5_0_6_3_37

eliotstock, 2019-10-22:
That was it, thank you! I'm sure I had this working even with SEC_MITM on SDK 14.2.0, but I could be mistaken.

closed 2019-10-22 by eliotstock