nordicsemi/Android-BLE-Library #133: why blemanager should work on mainthread?

szsheep, 2019-10-29:
Hi,
I used to adopt ble framework in new thread of fsm. I found that your framework is work on mainthread. what is the reason? Can I modify it to thread other than main thread?

thanks

philips77, 2019-10-29:
Hi,
So it has been found that the BLE works more stable when used from the main thread. But actually, I'm just starting to modify the library so the main thread will not be used that much, as it also introduces huge delays, a specially when someone wants to send a lot of data using Write Without Response. Jumping from thread to thread is slow, therefore the throughput drops drastically.
Please, expect a new release this or next week. I'll be publishing some 2.2-alphas to test on. 

I hope there will be no need to set the designated thread/handler. The intention is to make the lib easy to use and as stable as possible.

szsheep, 2019-10-30:
ok, however, since there are too much work processing on the main thread, would the ble manager be blocked by some tasks? as you said, main thread may be used too much. 

Have you do some test for how much lost of trasition efficiency  would be while the ble manager work on another thread instead of main thread?

thanks for your reply

philips77, 2019-10-30:
In my opinion the problem is on changing the thread, not that the main thread was selected. I need to debug it using a sniffer to compare.

philips77, 2020-03-10:
Btw, according to #159 `connectGatt(...)` needs to be called on the main thread. I think the lib should handle it, so this should be fixed before 2.2.0 goes final.

closed 2022-03-21 by philips77