nordicsemi/Android-BLE-Library #134: isRequiredServiceSupported() passed BluetoothGatt instance with no services

eliotstock, 2019-11-05:
Hi Aleksander. I'm finding that on connections to my peripheral that are not the very first one, but subsequent connections after bonding has already happened, the `BleManagerGattCallback.isRequiredServiceSupported()` method is passed a `BluetoothGatt` instance that at first appears to have none of my services on it. That causes my implementation of that callback to return false when it can't find any of the characteristics it needs. But the device is actually one of mine. I can tell from the BLE MAC address.

Although all of the `.getService()` calls to this `gatt` instance return null, if I set a breakpoint an inspect the `gatt` instance, it does have some services on it and those services do have characteristics on them. It's only the Generic Access and Generic Attribute services though, plus one of the Nordic ones.

I saw your comments on issue #17 and wondered if perhaps the service discovery was hitting a cache in the Android BLE stack, but the cache was dirty somehow. So I added a call to `shouldClearCacheWhenDisconnected()`, but it made no difference.

Here's my logcat:

```
2019-11-05 22:23:24.295 24465-24628/cc.biketracker.android D/BluetoothGatt: onClientRegistered() - status=0 clientIf=8
2019-11-05 22:23:24.298 24465-24465/cc.biketracker.android D/BluetoothAdapter: isLeEnabled(): ON
2019-11-05 22:23:24.384 24465-24628/cc.biketracker.android D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=8 device=C7:56:60:F0:56:58
2019-11-05 22:23:24.387 24465-24465/cc.biketracker.android D/BikeTrackerManager: onDeviceConnected()
2019-11-05 22:23:24.560 24465-24485/cc.biketracker.android W/tracker.androi: Accessing hidden method Ldalvik/system/CloseGuard;->warnIfOpen()V (light greylist, linking)
2019-11-05 22:23:25.154 24465-24628/cc.biketracker.android D/BluetoothGatt: onPhyUpdate() - status=0 address=C7:56:60:F0:56:58 txPhy=2 rxPhy=2
2019-11-05 22:23:26.019 24465-24465/cc.biketracker.android D/BluetoothGatt: discoverServices() - device: C7:56:60:F0:56:58
2019-11-05 22:23:26.031 24465-24628/cc.biketracker.android D/BluetoothGatt: onSearchComplete() = Device=C7:56:60:F0:56:58 Status=0
2019-11-05 22:23:26.046 24465-24465/cc.biketracker.android D/BikeTrackerManager: isRequiredServiceSupported()
2019-11-05 22:23:26.048 24465-24465/cc.biketracker.android W/BikeTrackerManager: Unsupported peripheral device. Some required services or characteristics were not found on device C7:56:60:F0:56:58.
2019-11-05 22:23:26.052 24465-24465/cc.biketracker.android W/BikeTrackerManager:   Found Bike Tracker service:               false
2019-11-05 22:23:26.056 24465-24465/cc.biketracker.android W/BikeTrackerManager:   Found Battery service:                    false
2019-11-05 22:23:26.059 24465-24465/cc.biketracker.android W/BikeTrackerManager:   Found TX Power service:                   false
2019-11-05 22:23:26.063 24465-24465/cc.biketracker.android W/BikeTrackerManager:   Found Nordic UART service:                false
2019-11-05 22:23:26.066 24465-24465/cc.biketracker.android W/BikeTrackerManager:   Found Battery level characteristic:       false
2019-11-05 22:23:26.069 24465-24465/cc.biketracker.android W/BikeTrackerManager:   Found TX Power level characteristic:      false
2019-11-05 22:23:26.072 24465-24465/cc.biketracker.android W/BikeTrackerManager:   Found Bike Tracker log characteristic:    false
2019-11-05 22:23:26.074 24465-24465/cc.biketracker.android W/BikeTrackerManager:   Found Bike Tracker unbond characteristic: false
2019-11-05 22:23:26.076 24465-24465/cc.biketracker.android W/BikeTrackerManager:   Found Nordic UART RX characteristic:      false
2019-11-05 22:23:26.079 24465-24465/cc.biketracker.android W/BikeTrackerManager: Instead, the device has the following service UUIDs:
2019-11-05 22:23:26.081 24465-24465/cc.biketracker.android W/BikeTrackerManager:   00001800-0000-1000-8000-00805f9b34fb
2019-11-05 22:23:26.083 24465-24465/cc.biketracker.android W/BikeTrackerManager:   00001801-0000-1000-8000-00805f9b34fb
2019-11-05 22:23:26.086 24465-24465/cc.biketracker.android W/BikeTrackerManager:   0000fe59-0000-1000-8000-00805f9b34fb
2019-11-05 22:23:26.088 24465-24465/cc.biketracker.android D/BikeTrackerManager: onDeviceNotSupported()
2019-11-05 22:23:26.088 24465-24465/cc.biketracker.android D/BikeTrackerManager: onDeviceDisconnecting()
2019-11-05 22:23:26.088 24465-24465/cc.biketracker.android D/BluetoothGatt: cancelOpen() - device: C7:56:60:F0:56:58
2019-11-05 22:23:26.091 24465-24628/cc.biketracker.android D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=8 device=C7:56:60:F0:56:58
2019-11-05 22:23:26.094 24465-24465/cc.biketracker.android D/BluetoothGatt: close()
2019-11-05 22:23:26.095 24465-24465/cc.biketracker.android D/BluetoothGatt: unregisterApp() - mClientIf=8
2019-11-05 22:23:26.095 24465-24465/cc.biketracker.android D/BikeTrackerManager: onDeviceDisconnected()
2019-11-05 22:23:26.107 24465-24465/cc.biketracker.android D/BikeTrackerManager: onDeviceDisconnected()
```

It's possible I'm using the library incorrectly. What do you suggest?

philips77, 2019-11-06:
Yes, that may be caching issue. Do you have Service Changed characteristic? Even if, Android still ignores it. On Android versions below 7 it was always caching, in 8-9 never caching, on 10 perhaps it's as it should be, I have to verify that.
If you can `shouldClearServicesWhenDisconnected()` the next connection should work. If you see 2 Generic and a Nordic one isn't if a dfu bootloader?

eliotstock, 2019-11-06:
Yeah, that was it! That last service it's showing is the Secure DFU service. It's stuck in the bootloader after a failed DFU. Flashing the device again to remove the bonds and DFU state fixed it.

I'll need to figure out why the DFU failed. Thanks.

closed 2019-11-06 by eliotstock