nordicsemi/Android-BLE-Library #135: WriteRequest write characteristic is too slow.

gksrb4, 2019-11-14:
I try to 
`final WriteRequest request = writeCharacteristic(mRXCharacteristic, data);				request.enqueue();`

[image]
It takes almost 100 ms to send data.
The connection interval is fast enough. how can I improve speed?

philips77, 2019-11-14:
Hi @gksrb4,
Yes, we are aware of this issue. It will be addressed in version 2.2. The problem is, that in the current version all BLE requests are performed from the main thread, while the gatt callbacks use their own thread, so there is a thread change before every write operation. For other operations this is acceptable, but for write without response this decreases the speed significantly. The new version will no longer use the main thread for everything, but I have to do a lot of testing, a specially on the older phones, before releasing it. The reason for using the main thread was that in the old days the BLE seemed to be more stable and there were less 133 errors popping up. But that also might have been a side effect of the delay caused be switching the thread.

gksrb4, 2019-11-14:
Okay, thanks to reply.

closed 2020-03-10 by philips77