nordicsemi/Android-BLE-Library #136: Quick disable and enable notify on the same characteristics in succession doesn't work

uberchilly, 2019-11-23:
When I enqueue disableNotifications and then enqueue enableNotifications in succession for the same characteristic I can see it does these operations on the embedded device but DataReceivedCallback doesn't receive any more data from that point. I also receive SuccessCallback for both disable and enable operations. 

More context: 
My plan is to make every fragment responsible for managing its own notifications to enable them on start and disable them on stop. I have 2 fragments, but 1 characteristic is listened on both fragments. That is how I get into the situation where I disableNotifications on the first one and very quickly afterwards enable it again on different screen because of how fragment lifecycle in android works when user switches screens. 

Any insight on why I don't receive data? 

[image]
[image]

philips77, 2019-11-23:
Perhaps the notification listener gets removed on disabling notifications? Food you try setting it again before enabling?

philips77, 2019-11-23:
Have a look at this line:
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/5e0e2f08c309a6de2376d9b8705c83f9e9a80d56/ble/src/main/java/no/nordicsemi/android/ble/BleManager.java#L2795

philips77, 2019-11-23:
To re-set the callbacks, set them in `done` after they are actually disabled.

uberchilly, 2019-11-25:
I have moved data callback into onDone callback, now it works. I am missing some data that is triggered immediately from the device though, but setting callback before enable and inside onDone fixes that issue as well. 
Tnx, closing the issue.

closed 2019-11-25 by uberchilly