nordicsemi/Android-BLE-Library #139: Library Using Procedure for a sample Read write operation.

VinayTShetty, 2019-12-13:
Hi thank you for providing library in frist hand.
i started working on android BLE projects.
i figured out the scanning ble library how to add via git and scan for the Devices.

But  when using this Android-BLE-Library i am getting to many gradle issues.
Not sure how to enable this library in android studio.
i try to De-bug nrftool box but its more and was not able to De-Bug it.

It would be helpful if i someone provide the source code of the app performing Read and write operation (which is built using Android-BLE-Library)

philips77, 2019-12-14:
Hi, there's nRF Blinky that's fairly simple.
What Gradle issues to you have?

VinayTShetty, 2019-12-15:
Hi philips,thanks for the Quick response thanks you..
 I already tried the nrf blinky code.
it s not scanning for all the Devices I have already raised the issue link(https://github.com/NordicSemiconductor/Android-nRF-Blinky/issues/34).NRF blinky app is scanning which I downloaded from play store if I remove UUID filtering from on optionItemSelected.
But the Githhub repository(https://github.com/NordicSemiconductor/Android-nRF-Blinky/issues/34)is not showing the devices when scanned

VinayTShetty, 2019-12-15:
Hi NordicSemiconductor/Android-BLE-Library,
I have already tried blinky example.
But I am facing some issues in that library too.
Its not scanning for the Devices issue link
<https://github.com/NordicSemiconductor/Android-nRF-Blinky/issues/34>.
The nrf blinky app downloaded from play store will scan for the Devices
when I disable UUID filtering from onoptionItemselected.
But the Github Repository
<https://github.com/NordicSemiconductor/Android-BLE-Library> code is not
scanning for the Device if I remove the UUID filtering from
onoptionItemselected.

philips77, 2020-03-11:
The BLE library is not intended to scan. Use native [BluetoothLeScanner](https://developer.android.com/reference/android/bluetooth/le/BluetoothLeScanner) or [Android Scanner Compat Library](https://github.com/NordicSemiconductor/Android-Scanner-Compat-Library) if your minimum sdk is below 21.

closed 2020-03-11 by philips77