nordicsemi/Android-BLE-Library #142: onDataReceived is not being called.

hasanmumin, 2019-12-28:
Hello,  I try to implement library between two android device, server-client, client sending message to server is fine but when server sending message is not being received by client. 
Here is my server:
```
public final static UUID DEVICE_UUID_SERVICE = UUID.fromString("7d2ea28a-f7bd-485a-bd9d-92ad6ecfe93e");
public final static UUID DEVICE_UUID_CHAR = UUID.fromString("7d2ebaad-f7bd-485a-bd9d-92ad6ecfe93e");
```
```
private void setupServer() {
        BluetoothGattService service = new BluetoothGattService(DEVICE_UUID_SERVICE,
                BluetoothGattService.SERVICE_TYPE_PRIMARY);
        BluetoothGattCharacteristic readCharacteristic = new BluetoothGattCharacteristic(DEVICE_UUID_CHAR,
                BluetoothGattCharacteristic.PROPERTY_WRITE |
                        BluetoothGattCharacteristic.PROPERTY_READ |
                        BluetoothGattCharacteristic.PROPERTY_NOTIFY
                ,
                BluetoothGattCharacteristic.PERMISSION_WRITE |
                        BluetoothGattCharacteristic.PERMISSION_READ
        );

        service.addCharacteristic(readCharacteristic);

        mGattServer.addService(service);
    }
```
```
private class GattServerCallback extends BluetoothGattServerCallback {
        @Override
        public void onConnectionStateChange(BluetoothDevice device, int status, int newState) {
            log("onConnectionStateChange " + device.getAddress()
                    + "\nstatus " + status
                    + "\nnewState " + newState);

            if (newState == BluetoothProfile.STATE_DISCONNECTED) {
                log("Device is disconnected." + device.toString());
                mHandler.post(new Runnable() {
                    @Override
                    public void run() {
                        removeDevice(device);
                    }
                });

            } else if (newState == BluetoothProfile.STATE_CONNECTED) {
                log("Device is connected." + device.toString());
                mHandler.post(new Runnable() {
                    @Override
                    public void run() {
                        addDevice(device);
                    }
                });
            }

        }

        @Override
        public void onCharacteristicReadRequest(BluetoothDevice device, int requestId, int offset, BluetoothGattCharacteristic characteristic) {
            log("onCharacteristicReadRequest "
                    + characteristic.getUuid().toString());

            sendResponse(device, requestId, BluetoothGatt.GATT_SUCCESS, 0, null);
        }

        @Override
        public void onCharacteristicWriteRequest(BluetoothDevice device, int requestId, BluetoothGattCharacteristic characteristic, boolean preparedWrite, boolean responseNeeded, int offset, byte[] value) {
            super.onCharacteristicWriteRequest(device, requestId, characteristic, preparedWrite, responseNeeded, offset, value);
            if (responseNeeded) {
                sendResponse(device, requestId, BluetoothGatt.GATT_SUCCESS, offset, value);
            }
        }

        @Override
        public void onNotificationSent(BluetoothDevice device, int status) {
            log("onNotificationSent " + status);
        }
    }
    public void sendResponse(BluetoothDevice device, int requestId, int status, int offset, byte[] value) {
        mHandler.post(() -> mGattServer.sendResponse(device, requestId, status, 0, null));
    }
```
Message send code:
```
public void sendMessageToDevice(BluetoothDevice device, String message) {
        BluetoothGattService service = mGattServer.getService(DEVICE_UUID_SERVICE);
        BluetoothGattCharacteristic characteristic = service.getCharacteristic(DEVICE_UUID_CHAR);
        characteristic.setValue(message.getBytes(StandardCharsets.UTF_8));
        mGattServer.notifyCharacteristicChanged(device, characteristic, false);
  }
```

onNotificationSent is being called with 0 which is mean GATT_SUCCESS

----------------------------------------------------------------------

Here is my client:

```
public class DeviceManager extends BleManager<BleManagerCallbacks> {

    private final Logger LOGGER = Slf4jHelper.getLogger(this.getClass().getName());

    private final DeviceDataCallback mDeviceDataCallback = new DeviceDataCallback() {
        @Override
        public void onDataReceived(@NonNull BluetoothDevice device, @NonNull Data data) {
            toast("onDataReceived: " + new String(data.getValue()));
        }

        @Override
        public void onDataSent(@NonNull BluetoothDevice device, @NonNull Data data) {
            toast("onDataSent: " + new String(data.getValue()));
        }

        @Override
        public void onInvalidDataReceived(@NonNull BluetoothDevice device, @NonNull Data data) {
            toast("onInvalidDataReceived: " + new String(data.getValue()));
        }
    };
    private boolean mSupported;
    private BluetoothGattCharacteristic mDBluetoothGattCharacteristic;
    private final BleManagerGattCallback mGattCallback = new BleManagerGattCallback() {

        @Override
        protected void initialize() {
            setNotificationCallback(mDBluetoothGattCharacteristic).with(mDeviceDataCallback);
            readCharacteristic(mDBluetoothGattCharacteristic).with(mDeviceDataCallback).enqueue();
            enableNotifications(mDBluetoothGattCharacteristic).enqueue();
        }

        @Override
        protected boolean isRequiredServiceSupported(@NonNull BluetoothGatt gatt) {
            final BluetoothGattService service = gatt.getService(NearbyActivity.DEVICE_UUID_SERVICE);
            if (service != null) {
                mDBluetoothGattCharacteristic = service.getCharacteristic(NearbyActivity.DEVICE_UUID_CHAR);
            }

            boolean writeRequest = false;
            if (mDBluetoothGattCharacteristic != null) {
                final int rxProperties = mDBluetoothGattCharacteristic.getProperties();
                writeRequest = (rxProperties & BluetoothGattCharacteristic.PROPERTY_WRITE) > 0;
            }

            mSupported = mDBluetoothGattCharacteristic != null && writeRequest;
            return mSupported;
        }

        @Override
        protected boolean isOptionalServiceSupported(@NonNull BluetoothGatt gatt) {
            return !mSupported;
        }

        @Override
        protected void onDeviceDisconnected() {
            mDBluetoothGattCharacteristic = null;
        }
    };

    /**
     * The manager constructor.
     * <p>
     * After constructing the manager, the callbacks object must be set with
     * {@link #setGattCallbacks(BleManagerCallbacks)}.
     * <p>
     * To connect a device, call {@link #connect(android.bluetooth.BluetoothDevice)}.
     *
     * @param context the context.
     */
    public DeviceManager(@NonNull Context context) {
        super(context);
    }

    @Override
    public void log(int priority, @NonNull String message) {
        LOGGER.info(message);
    }

    private void toast(String message) {
        Toast.makeText(getContext(), message, Toast.LENGTH_SHORT).show();
        LOGGER.info(message);
    }

    @Override
    protected boolean shouldClearCacheWhenDisconnected() {
        return true;
    }

    @NonNull
    @Override
    protected BleManagerGattCallback getGattCallback() {
        return mGattCallback;
    }

    public void send(final String message) {
        // Are we connected?
        if (mDBluetoothGattCharacteristic == null)
            return;

        writeCharacteristic(mDBluetoothGattCharacteristic, Data.from(message))
                .with(mDeviceDataCallback).enqueue();
    }
}
```
client Activity, I am using Android-Scanner-Compat-Library for scanning:
```
private void connect() {
        mDeviceManager = new DeviceManager(getApplication());
        mDeviceManager.setGattCallbacks(this);

        mDeviceManager.connect(mDeviceServer)
                .retry(3, 100)
                .useAutoConnect(false)
                .enqueue();
    }
```

Do I miss something ?
Server is galaxy s9 plus with android 9, client is  android 5.1. 
Thank you.

philips77, 2019-12-28:
Hello,
3 things:
1. You're missing Client Configuration Characteristic Descriptor that the client can use to enable notifications. Add it to your server characteristic, the one with notify property.
2. I'd call `sendResponse` from the same thread, but with a help of a handler.
3. I'm working on adding server support to the ble library right now. In Monday you should be able to pull me changes. It is to be used just like the client, work different requests from the `BleManager` class.

hasanmumin, 2019-12-28:
@philips77 Thank you. I am going to follow your things.

hasanmumin, 2019-12-28:
@philips77 Thank you, with 'CLIENT_CHARACTERISTIC_CONFIG_DESCRIPTOR_UUID' working fine.

hasanmumin, 2019-12-29:
@philips77  Hello again, unfortunately now client getting error when try to send message to server '/BleManager: onDescriptorWrite error 133'
I created a new repository which is simulating my scenario 
Server changes :
https://github.com/hasanmumin/android-ble/blob/master/Server/app/src/main/java/com/hasanmumin/ble/server/ServerActivity.java#L102
Client Changes:
https://github.com/hasanmumin/android-ble/blob/master/Client/app/src/main/java/com/hasanmumin/ble/client/ClientManager.java#L34

Here is my full repository:
https://github.com/hasanmumin/android-ble
 Thank you.

hasanmumin, 2019-12-29:
I had to implement onDescriptorWriteRequest and onDescriptorReadRequest.
Thank you! working fine now!

closed 2019-12-29 by hasanmumin

philips77, 2019-12-30:
Hi, have a look at #143