nordicsemi/Android-BLE-Library #146: NRF ToolBox UART De-Bugging.

StudyMaterialsAndProjects, 2020-01-06:
Hi ,
frist of all thanks for providing the source for this application.[NRF Toolbox](https://github.com/NordicSemiconductor/Android-nRF-Toolbox).

Here i am De-Bugging the application for the UART in that i am seeing a Connection is done using
Google API.(Screen Shot i have attached).
[image]

But the documentation says that UART is using below code to do the connection.[Link](https://github.com/NordicSemiconductor/Android-BLE-Library)
[image]

which is the actually right...?

can you please provide the code where the BLE library is used...
so that it will be easy to De-Bug.

philips77, 2020-01-06:
Have a look at UARTManager`` class. The Google API here is used to connect to Wear devices, which can also be used to send UART commands in nRF Toolbox. This feature is optional and we are not sure if it still works on Google Wear watches, as the sample app we made is very old now.

VinayTShetty, 2020-01-06:
Hi philips,
i want to connect,read write and notify data from BLE device.
i am looking for a simple example.which uses **Android-BLE-Library**,can you please tell me where i can get this..
Thanks in Advance.

philips77, 2020-01-06:
Have a look at [nRF Blinky](https://github.com/NordicSemiconductor/Android-nRF-Blinky).

closed 2020-03-11 by philips77