nordicsemi/Android-BLE-Library #157: BleManager calls internalConnect after bond is lost v2.1.1

SarthakWhoop, 2020-02-06:
Connect a BLE device by setting the `autoConnect` flag to `True` and then after the device is bonded successfully, delete the bond from the Android Settings ("forget device"). `BleManager` then immediately connects to the same device again:
`					if (wasConnected && mInitialConnection) {
						internalConnect(gatt.getDevice(), null);
					} `
This prevents users from removing the bond for the device because the `BleManager` will reconnect and repair to the same BLE device once the bond is lost.

philips77, 2020-02-07:
What would be the desired behavior?

SarthakWhoop, 2020-02-07:
I think the desired behavior should be to **NOT** connect/pair to the same BLE device again and to release the GATT resources. If a user deletes the bond from the Android device settings, it should not reconnect back to the same device again since it is against what the user wanted in the first place.

prfarlow1, 2020-02-13:
@philips77 Taking a step back from the specific code flows in this issue: What is the best way for an app using the Nordic BLE Library to determine if a disconnection happened because the user removed the device bond from the Android settings Bluetooth menu, compared to something outside of the software's control (like the device moving out of range)

philips77, 2020-03-11:
Hello, sorry for the delay. I was buried with another tasks.
First of all, the library works also with not bonded devices, so the solution proposed by @SarthakWhoop above would not work. But I close the connection when bond state changed from BONDED to NONE. This should solve your issue and should have no impact on other situations.

closed 2020-06-30 by philips77