nordicsemi/Android-BLE-Library #158: two onDeviceReady callbacks after install of Jan 1, 2020 security patch

RayPelletier, 2020-02-15:
On one of test devices my app is receiving two onDeviceReady callbacks. They come in quick succession and only the second accurately reflects the device's readiness. I've found no way to distinguish between the two. This behavior only started happening after installing the Jan 1, 2020 security patch. Connect is only called once. 

I'm using version 2.1.1 of the library
The test device is a Samsung S8 running Android 10

philips77, 2020-06-24:
Hello,
Sorry for such long delay. In the meantime we released version 2.2.0. Did you have time to test it?

RayPelletier, 2020-06-24:
We will be returning to Bluetooth in about a week. I will update the thread
with results.

philips77, 2022-03-21:
I'm closing the issue due to inactivity.

closed 2022-03-21 by philips77