nordicsemi/Android-BLE-Library #159: onConnectionStateChange() not called on Huawei P20 lite (Alpha06)

balazs-muller, 2020-02-19:
Hello,

We using 2.1.1 version of this library and it is work perfectly, Thanks!
But, we started testing 2.2.0-alpha06 version, and one of phone type (**Huawei P20 lite**, Android 9) has a problem.
We **cannot connect** to devices, **because** the **_BluetoothGattCallback.onConnectionStateChange()_ never called**. And the connection always drops _FailCallback.REASON_TIMEOUT_.

I try to find the reason.
I take a method break point into _BleManagerHandler.BluetoothGattCallback_ _onServicesDiscovered_ and the _GattCallbacks_ like _onConnectionStateChange()_ called normally (but slower because the method breakpoint slows down the threads. -> I think this is a thread issue.)
So, I move the device.connectGatt() method to main thread _(runOnUiThread)_ in BleManagerHandler.internalConnect() and the connections work well.

I found interesting things:
- If no Code change just use method breakpoints:
After onConnectionStateChange() called with Connected state, called the 
_Connection parameters updated (interval: 7.5ms, latency: 0, timeout: 5000ms)_
and (~1300 ms) later 
_Connection parameters updated (interval: 50.0ms, latency: 0, timeout: 5000ms)_
and (~3000 ms) later 
_Connection parameters updated (interval: 30.0ms, latency: 0, timeout: 4000ms)_
and than Services discovered and connection works well.

- if i call device.connectGatt() method on main thread:
After _onConnectionStateChange()_ called with Connected state, called the 
_Connection parameters updated (interval: 7.5ms, latency: 0, timeout: 5000ms)_

maybe related issues:
- [https://github.com/NordicSemiconductor/Android-BLE-Library/issues/78](url)
- [https://github.com/NordicSemiconductor/Android-BLE-Library/issues/135](url)

Maybe the problem is in the phone's Bluetooth stack itself. 

Do you have any idea?
Thanks!

philips77, 2020-03-10:
The connection parameters update is caused by speeding up the service discovery. Android switches to 7.5ms interval to make the discovery as fast as possible, and then back to a default interval on one requested by the peripheral.

Regarding threading issue, in 2.1.1 all BLE commands were called from the main thread in inside the library, even when called from another thread. In 2.2.0 they are called from the caller thread, and following ones (if queued) from ble thread. I have never tried connecting from a different thread than the main, but non-ui threads had issues before, that's one of the reasons why 2.1.1 was always switching to the main thread. I don't know the rot cause of this issue.

philips77, 2020-06-24:
Hello, can we close this issue?

okbean, 2020-12-15:
I met this issue on other phone . ConnectGatt must be called  in ui thread on some kind of phones。In Android-BLE-Library ，BleManagerHandler::enqueue can be called in worker thread, BluetoothGattCallback::onConnectionStateChange is also called in binder thread . so this two  condition could cause this issue!

philips77, 2022-03-21:
I'm closing the issue due to inactivity.

closed 2022-03-21 by philips77