nordicsemi/Android-BLE-Library #160: Something wrong with writing long bytes.

onatakduman, 2020-02-26:
Hi, 
I am working with FCS-BT630 (uses nRF 52832 chip) module and I am trying to sending long bytes (mp3 file).
MTU size is 247.
Phone: POCOPHONE F1

```java
public void sendFile(byte[] file) {
        writeCharacteristic(mCharacteristicWrite, file)
                .with(mDataCallback)
                .split(mDataCallback)
                .enqueue();
}
```
```java
        requestMtu(247)
                    .with((device, mtu) -> log(Log.INFO, "MTU set to " + mtu))
                    .fail((device, status) -> log(Log.WARN, "Requested MTU not supported: " + status)).enqueue();
```
```
2020-02-26 15:58:56.917 E/FILE SIZE: 37332 byte
. . .
2020-02-26 15:59:16.202 E/TRANSFER START: 15:59:16.193
2020-02-26 15:59:16.258 I/MyBleManager: onPacketSent: Index: 0, Data: 244
2020-02-26 15:59:16.293 I/MyBleManager: onPacketSent: Index: 1, Data: 244
2020-02-26 15:59:16.325 I/MyBleManager: onPacketSent: Index: 2, Data: 244
2020-02-26 15:59:16.367 I/MyBleManager: onPacketSent: Index: 3, Data: 244
2020-02-26 15:59:16.394 I/MyBleManager: onPacketSent: Index: 4, Data: 244
2020-02-26 15:59:16.424 I/MyBleManager: onPacketSent: Index: 5, Data: 244
. . .
2020-02-26 15:59:20.901 E/TRANSFER FINISH: 15:59:20.896
2020-02-26 15:59:20.902 E/TRANSFER TOTAL TIME: 4.703 s
2020-02-26 15:59:20.907 E/TRANSFER FILE: 7,75 KBps
```

When I measured the speed rate is 7,75 KB/s as seen. Expected speed is between 65 - 100 KB/s. 

- What could be the main cause of the condition? 
- How can I increase the speed?

I tried several method for splitting and sending long bytes but I can not able to reach expected transfer rates.

philips77, 2020-02-27:
Hello,
Make sure the characteristic you're writing is using Write Without Response (aka Write Command). Using this write type multiple packets can be sent in a single connection interval. Also, the speed depends on MTU (I see you have it set to 247, which is very good), DLE (Data Length Extension) - newer phones running Android 6 and newer support this Bluetooth 4.2 feature and on how many packets your phone can send and the device can receive in a single connection interval. It can be easily seen using a sniffer, if you have one.You can also use [nRF Sniffer](https://www.nordicsemi.com/Software-and-tools/Development-Tools/nRF-Sniffer-for-Bluetooth-LE/Download#infotabs) from Nordic, which is free and is using Wireshark. In wireshark you may use Menu Statistics -> IO Graph to display packets on the graph.

onatakduman, 2020-02-27:
Thanks for reply, I check Write Type and changed it to Write Without Response so I get expected speeds but

philips77, 2020-02-29:
Are you using the BLE library in version 2.2.0-alpha06? It has improved transfer speed a bit, as ours not switching threads.
There's no way to make the transfer faster, the lib is trying to send as fast as possible. You can only slow it down if needed.
Regarding missing bytes: BLE is a reliable protocol, so each packet is retransmitted if missed, but as it's write without response, they may be lost on the firmware side after receiving in the stack, before passing to the app. The only way to improve it is by adding some app level validation, like crc, etc (flow control).

philips77, 2020-03-10:
Hello, any updates on this?

closed 2020-03-11 by philips77

onatakduman, 2020-03-12:
There is no any update, thanks for guidance.