nordicsemi/Android-BLE-Library #163: Initialize app problem

NimaaZx, 2020-03-08:
Hello. I wanted to implement this lib from example guide. I tried to define a class that you wrote in README file . at first I wanted to convert them to kotlin. because I started android programming with kotlin. I went with java. but some variables are not recognized at all. what do i do ? 
I write all of them for you:

dataCharacteristic in :
```
 if (firstCharacteristic != null) {
                final int properties = dataCharacteristic.getProperties();
                notify = (properties & BluetoothGattCharacteristic.PROPERTY_NOTIFY) != 0;
            }
```

controlPointCharacteristic in :

```
      final int properties = controlPointCharacteristic.getProperties();
             writeRequest = (properties & BluetoothGattCharacteristic.PROPERTY_WRITE) != 0;
```

callback in :
`            setNotificationCallback(firstCharacteristic, callback);`

philips77, 2020-03-09:
The sample in Readme uses an example service and characteristics. You need to change them to your in the device you connect to, assuming you do want to connect to the device.
Use nRF Connect app to find your what services your wrist band has and use those uuids in your app, if you need to send it read any date other than rssi.

philips77, 2020-03-09:
Have a look at nRF Blinky sample app on GitHub. You'll find there BlinkyManager.

NimaaZx, 2020-03-09:
> Have a look at nRF Blinky sample app on GitHub. You'll find there BlinkyManager.

Oh. Thank you for your concern. I do not want  connect to my ble wristband. I just what scanning, filters, do some calculations. 

1- You mean I should implement something that I need not everything?

Yesterday I could use own default android ble library for doing that things I wanted. 

2-Could you tell me what is the difference between NordicSemiConductor library and that default library in android ? 
What capabilities or extras...

philips77, 2020-03-09:
For scanning only, you should use the scanner API, not this library. If you want your app to support Android 4.3+, not only the latest Android versions, we recommend checking on https://github.com/NordicSemiconductor/Android-Scanner-Compat-Library. The scanner compat lib tries to emulate the new features and API on older Android versions, so you won't need to write duplicate code (scanner API has changed in Android 5 and was extended later).

The BLE Library is a handy lib for handling connections. It has more readable API than the native one, handles some common errors and separates Bluetooth low-level implementation from the high-level device API.

closed 2020-03-10 by philips77