nordicsemi/Android-BLE-Library #165: On Devices like Samsung S10e, Asus Zenfone Max BLE connection not working

pravingaikwad123, 2020-03-13:
On some devices the BLE connection is not woring properly.
in gattCallback > onConnectionStateChange we are receiving state as BluetoothGatt.GATT_SUCCESS and status as BluetoothProfile.STATE_CONNECTED but then it fails to automatically.
One common thing that I have observed is BLE works perfectly fine when we restart the phone. But as soon Bluetooth is turned on-off then we start seeing this behaviour.
Also, I think there is some issue with clientif attribute in gatt as whenver connection fails its value is greater than 7.
Any solution for this ?

philips77, 2020-03-13:
Looks like the resources are not released properly. I've heard that `gatt.disconnect()` method should be called even if not connected, `gatt.close()` is not enough. I'll look into this before releasing 2.2.0 final version.

pravingaikwad123, 2020-03-13:
Even I tried to use other libraries and referred to Android's BLE
documentation.
I have posted similar issue on StackOverflow as well :
https://stackoverflow.com/q/60666682/5582162
Hope it helps.

philips77, 2022-03-21:
I'm closing the issue due to inactivity.

closed 2022-03-21 by philips77