nordicsemi/Android-BLE-Library #166: List of Coded PHY Supported/Tested Phones

kiafaldorius, 2020-03-14:
Does anyone happen to know if there's a full list somewhere with compatible Coded PHY supported phones?

I tried it on a Pixel 4 which is supposed to support it, and claims it supports it via `isLeCodedPhySupported`, but I've been unable to get any advertisements. I even tried using the nrfConnect app, and getting the same result.

It works on the Samsung S10+. Can anyone point me to what I can look into to debug this?

philips77, 2020-03-16:
There's number of phones that support Coded phy, but a device needs to advertise on LE 1M as Primary phy to be found, which completely brakes the purpose of long range. Pixel is this group. Actually, we only know only One Plus 6 that can scan packets with Primary phy Coded. Good to know S10+ is also capable to do so.

kiafaldorius, 2020-03-20:
Dang, that sucks because the Pixel itself can advertise on LE Coded in both the primary and secondary channels. Why would it only support scanning on LE 1M?!?

So to re-iterate the first question...is there a list somewhere? If we don't have it, can we start compiling it?

philips77, 2020-03-20:
So far I know only One Plus 6 and Samsung S10(+) that can scan using primary PHY Coded.
I'm working on a test feature in nRF Connect for Android that will let us create such list, but this is still month before releasing.

kiafaldorius, 2020-03-20:
Ok, thank you! I'll keep my eye out for it.

kiafaldorius, 2020-05-13:
I can now confirm that the One Plus 7 series can also scan using primary PHY Coded.

And can also confirm that Google Pixel 4 won't support primary PHY Coded: https://issuetracker.google.com/issues/148487283

simicNavcast, 2021-06-11:
Here are some devices I have tested and confirmed to have BT5 Code PHY Long range.

Galaxy Tab A7
Galaxy S10 and above
Oneplus 7 and above

Somebody above said Oneplus 6 works, but if I remember correctly, I think it didn't work for us when a colleague tested it. It might depend on which version of Oneplus 6 it is because there are sometimes differences between versions sold in different countries.

Some devices possibly have it according to the Bluetooth ICS page, but they usually come with this comment:

C.2: Excluded IF SUM ICS 21/14 “Core v4.2” is supported, otherwise Optional.

Some devices I tested that potentially had it according the ICS page, actually didn't have it or it didn't work for scanning. In the menu of the nRF connect app, you can tap "Device information" and it will have YES or NO for "Long range (PHY Coded) supported".

For the devices I tested that potentially had it but didn't work, they failed at different points. Some showed NO in nRF connect for "Long range (PHY Coded) supported", others like LG K31 said YES, but when I tried to scan, it couldn't see a Code PHY advertisement. Xiaomi POCO X3 NFC is supposed to have it according to ICS, but a colleague of mine just tested it and it wasn't seeing Long range advertisements. I asked him to look in nRF connect to see if it says YES for "Long range (PHY Coded) supported" but I haven't heard back yet.

These devices potentially have it according to the Bluetooth ICS page, but I doubt it works (at least not for scanning) with any of the LG or Xiaomi phones and don't have high confidence it will work on the Motorola phone either.

LG Velvet
LG K61
LG K41S
LG K300
 
Xiaomi Redmi 9 and 9C
Xiaomi Redmi Note 9
Xiaomi Mi 9
Xiaomi Mi 10T Lite and Pro
 
Motorola Edge +

A commenter above mentioned that some phones can communicate with Code PHY, but can only scan in 1M. I didn't even test that out because for my purposes, those phones would be useless. I need them to be able to scan in Long Range because we are actually working with Long Range requirements.

gamelaster, 2021-06-17:
Samsung Galaxy S21 works absolutely fine.
To opposite side, OnePlus 6T don't see secondary advertisement channel, and it's difficult to connect into coded-enabled peripheral (although, it's possible)

mafaneh, 2021-09-12:
This is a great thread! I appreciate the efforts of coming up with a list of phones that support Coded PHY. I have a related question (and I apologize if this is not a bit tangential to the topic), but I honestly didn't know where else to ask the question.

I am currently using a OnePlus Nord N10 5G and I can confirm it works well for scanning, advertising (primary & secondary), and connecting all using Coded PHY.

The issue I am running into is I'm not able to scan **only** Coded PHY advertisements. According to the Android API documentation, you can set the PHY used for scanning to be either 1M + Coded PHY, 1M only, or Coded only. However, I am still receiving non-Coded PHY advertisements even when I set the Phy to Coded PHY using the .setPhy(BluetoothDevice.PHY_LE_CODED) while setting the Scan Settings. The codebase I am using is the Android-nRF-Blinky GitHub repo.

Please let me know if I should be posting this somewhere else. I'll delete and post where appropriate!

kboiss, 2021-11-24:
We have tested the Samsung M51 to work

Not Working are the Samsung A32 5G

kiafaldorius, 2021-12-13:
I can confirm the Samsung A32 5G does not pick up primary LE_Coded only advertising payloads.

The issue mentioned on the OnePlus 6T happens with a few other phones. It gets worse if you increase the advertising interval.

@mafaneh That's a different issue, open a new ticket for that if you need help please. Short answer: Android doesn't let you scan using LE Coded only. It's only 1M+LE_CODED or 1M-only.

chemary, 2022-08-17:
My Samsung S20 was able to scan LE_CODED advertisements perfectly until the last 2022-August security update, in some pages it talks about this update adding improvements to the security of Wifi and Bluetooth. That's sad I may try to revert this update but seems not trivial, I used nRF Connect for Android to scan for some devices I was working on that only advertise on LE_CODED.

philips77, 2022-08-26:
Samsung Galaxy A8 (SM-X200, Android 11) tablet can scan packets sent on Primary and Secondary LE Coded.
Lenovo M10 (TB-X306F, Android 10) tabled cannot.

dsconyers, 2022-10-09:
Cat S42 (Android 11) supports the LE Coded PHY, but it can't scan for LE_CODED advertisements.

OnPl10, 2023-06-07:
OnePlus 10 Pro 5g has coded PHY support but I have no idea how to use it to find my earbuds. Anyone have some step by step instructions for dummies?

s3ni0r, 2023-08-14:
Hello, what about smartwatches like Samsung smartwatch 6? how can one check if they support coded phy?

rajeevinfolitz, 2024-05-10:
Did anyone find a compiled list of devices that supports CODED PHY

msimic777, 2024-05-10:
This forum topic seems to be the only compiled list that exists on the internet.

sameer, 2024-10-24:
This does not appear to have been fixed, at least as of Android 15 build number AP3A.241005.015.

paulchaffey, 2024-10-24:
> This does not appear to have been fixed, at least as of Android 15 build number AP3A.241005.015.

Hi Sameer, 
where did you find this ? I have spent some time looking through the Android 15 API and I don't know where I should be looking - I'm a HW Engineer, so I am not so up to speed in this area.
I'm also very confused when a bug is marked as fixed, how you know which release it will be in.
Thanks, Paul

sameer, 2024-10-25:
> Hi Sameer,
> where did you find this ? I have spent some time looking through the Android 15 API and I don't know where I should be looking - I'm a HW Engineer, so I am not so up to speed in this [...]

Hi Paul, this is based on testing with 2 Android 15 phones. Scanning from device A still only finds 1M PHY advertisements made by device B. Advertisements on 2M + Coded are not found.

philips77, 2024-10-25:
I think that's hardware dependent. There are phones with older Android versions that can scan Coded PHY as primary one.

sameer, 2024-10-25:
> I think that's hardware dependent. There are phones with older Android versions that can scan Coded PHY as primary one.

Based on the linked issue tracker, this appears to be a bug introduced in some new-ish version of Android. Is that not the case?

paulchaffey, 2024-10-25:
I have a pixel 6 running Build number AP41.240925.009. I guess I can check it, but which App is a good question ?

paulchaffey, 2024-10-25:
> I have a pixel 6 running Build number AP41.240925.009. I guess I can check it, but which App is a good question ?

Would Nordic's nRF connect be up to date enough ? I would guess the API is the same, but the scanning for long range PHYs is connected up underneath correctly - or am I over simplifing things ?

philips77, 2024-10-25:
If the phone reports LE Coded advertising, nRF Connect will show it on the Scanner tab.

SimEyeSee, 2024-10-26:
The only real way to check it is to try to scan for a device that you know advertises Bluetooth LE Coded. The problem with using nRF Connect is that some phones will report yes for LE Coded, but they can only connect with an LE Coded device, they can't scan for one. So the device would have to advertise using 1M or 2M, but then connect using Coded.

msimic777, 2024-10-26:
The only real way to check it is to try to scan for a device that you know advertises Bluetooth LE Coded. The problem with using nRF Connect is that some phones will report yes for LE Coded, but they can only connect with an LE Coded device, they can't scan for one. So the device would have to advertise using 1M or 2M, but then connect using Coded.

philips77, 2024-10-27:
Yes. From our experience, all that claim support for LE Coded can advertise with such. Only few can scan when Primary PHY is Coded, which is the main point of this PHY for long range use case.

fanoush, 2025-05-06:
Just FYI Motorola Edge 30 with Android 12 can scan and connect to devices advertising over coded phy only. And BTW, we just added coded phy connections (both central and peripheral) and advertising to Bangle.JS 2 watch if you want  a device to test. I have tested it with this phone (and also with some other nrf52840 boards running Espruino).

marisol-valencia, 2025-06-04:
Im conducting some market research to guide the development of new features. Our product already has BLE capable chips integrated in the hardware and the long range could be a nice breakthrough if customers had access to these channel types. Does anyone have insight about the evolution of market share for mobile devices that will support BLE long range ie coded phy.

noffeperfekt, 2026-10-29:
I can confirm that Nothing phone 3a has full Bluetooth Low Energy support, including PHY Coded, extended advertising with a maximum advertising length of 1650