nordicsemi/Android-BLE-Library #169: Cannot be used with Kotlin?

travisgriggs, 2020-03-20:
I'm trying to create a ValueChangedCallback class in Kotlin, e.g.

philips77, 2020-03-21:
This is by design. You should use methods of BleManager to create this object:
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/e95a814f4c0ecd08a75e5f0e53c68e2b8dbdf70c/ble/src/main/java/no/nordicsemi/android/ble/BleManager.java#L610
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/e95a814f4c0ecd08a75e5f0e53c68e2b8dbdf70c/ble/src/main/java/no/nordicsemi/android/ble/BleManager.java#L628
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/e95a814f4c0ecd08a75e5f0e53c68e2b8dbdf70c/ble/src/main/java/no/nordicsemi/android/ble/BleManager.java#L644
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/e95a814f4c0ecd08a75e5f0e53c68e2b8dbdf70c/ble/src/main/java/no/nordicsemi/android/ble/BleManager.java#L660

philips77, 2020-03-21:
Also, this library is fully compatible with Kotlin.

closed 2020-03-21 by philips77