nordicsemi/Android-BLE-Library #184: onDeviceDisconnected callback reason is incorrect

fuhaodev, 2020-04-03:
When the disconnect method is called, the reason of the onDeviceDisconnected callback is 4, which is NOT SUPPORTED, and should be 0？

philips77, 2020-04-03:
Are you testing the alpha13 version?

philips77, 2020-04-03:
OK, but I think I know where the issue is. Thank you for the bug report!

philips77, 2020-04-03:
I think in all the previous versions there was an issue, where even in case of a non supported device, the failure reason in `ConnectRequest#fail(...)` was `FailCallback.REASON_DEVICE_DISCONNECTED` instead of `FailCallback.REASON_DEVICE_NOT_SUPPORTED`. At least by looking at the code.

philips77, 2020-04-03:
This commit has change order of calls and introduced this issue: https://github.com/NordicSemiconductor/Android-BLE-Library/commit/0555ae30c8b11619471b97a3904115142102dba5
In one of the 2.2.0-alphas I spotted it and added this temporary variable `notSupported`, but this is never cleared when the device actually is supported and disconnects. Thank you again!

philips77, 2020-04-03:
Fixed in 2.2.0-beta01. Could you confirm?

closed 2020-04-03 by philips77

fuhaodev, 2020-04-04:
It's ok in 2.2.0-beta01, thanks.