nordicsemi/Android-BLE-Library #186: ConnectionPriorityRequest problem

moritz-eder, 2020-04-15:
Hi all,
we found an issue with requesting high connection priority on newer Android devices. Primarily on Samsung devices.
Reproducible on Samsung Galaxy XCover Pro or Samsung Galaxy A10.

The issue is the following:
- Request High priority connection with BleManager.requestConnectionPriority(ConnectionPriorityRequest.CONNECTION_PRIORITY_HIGH) and add callbacks for done and/or fail and enqueue, or do it synchronous with await()
- The callbacks will never be called back/the await call will never return.

Not sure why this is, but it seems as if the onConnectionUpdated() method in BleManagerHandler never gets called and thus the request does not succeed or fail ever.

Our solution currently is to answer the request just like it is done in between SDK versions Lollipop and Oreo, by calling back with success shortly after successfully sending the request.

I would be super thankful if you have a better fix, than just ignoring the callbacks.

moritz-eder, 2020-04-16:
I also tried out making the ConnectionPriorityRequest a Timeoutable Request, which seems to work fine for us as well and could maybe even be a general solution...

philips77, 2020-04-20:
Hi,
What's the Android version on your phone?

moritz-eder, 2020-04-20:
Hi.
we see it happening on a number of devices. For example on Samsung A20 (Androi 9) or XCover Pro (Android 10). But it does not happen on Huawei P20 (also Android 9)

philips77, 2020-05-07:
Hello,
Have a look at the commit. The Connection Priority request should now succeed at most after 200 ms on any phone. If the native callback wasn't called until then, it will be called by a timer.

closed 2020-06-24 by philips77