nordicsemi/Android-BLE-Library #188: Crash in BleManager.onDeviceReady (BleManager.java:2215)

valeriyo, 2020-04-25:
With v2.1.1, got a crash once on Samsung Galaxy S9 (Android 10) because `mBluetoothGatt` was `null`:

```
		protected void onDeviceReady() {
			mCallbacks.onDeviceReady(mBluetoothGatt.getDevice());
		}
```

Callstack:
```
Fatal Exception: java.lang.NullPointerException: Attempt to invoke virtual method 'android.bluetooth.BluetoothDevice android.bluetooth.BluetoothGatt.getDevice()' on a null object reference
       at no.nordicsemi.android.ble.BleManager$BleManagerGattCallback.onDeviceReady(BleManager.java:2215)
       at no.nordicsemi.android.ble.BleManager$BleManagerGattCallback.nextRequest(BleManager.java:3158)
       at no.nordicsemi.android.ble.BleManager$BleManagerGattCallback.onServicesDiscoveredSafe(BleManager.java:2613)
       at no.nordicsemi.android.ble.MainThreadBluetoothGattCallback.lambda$onServicesDiscovered$1$MainThreadBluetoothGattCallback(MainThreadBluetoothGattCallback.java:112)
       at no.nordicsemi.android.ble.-$$Lambda$MainThreadBluetoothGattCallback$js3Daz29T8D5sg6HeYmRg0i-uuU.run(:6)
       at android.os.Handler.handleCallback(Handler.java:883)
       at android.os.Handler.dispatchMessage(Handler.java:100)
       at android.os.Looper.loop(Looper.java:237)
       at android.app.ActivityThread.main(ActivityThread.java:7811)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:493)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1076)
```

This looks like a race condition between the main thread runnable and `close()` method.

philips77, 2020-04-25:
Could you try 2.2.0-beta? Some API has changed a bit, but should not be difficult to migrate. And I think it's more stable.

philips77, 2020-06-24:
Hello,
Any progress in this issue? Did you try version 2.2.0?

philips77, 2020-06-30:
Duplicate of #156.

closed 2020-06-30 by philips77

valeriyo, 2020-08-13:
@philips77 we're shipping with version 2.2.3 now - so far so good. Will let you know if I see it. If I don't follow up - consider it fixed. Thanks!