nordicsemi/Android-BLE-Library #189: Identify Maximum connection limit reached on connecting via Android BLE

jainsameeksha, 2020-04-28:
I am developing using Android BLE want a specific error code that could help me determine the reason of connection failure while connecting to a device, specifically the scenario i want to cover is for maximum connections limit being reached.
I want any specific error code or method to identify that connecting to a BLE device failed due to maximum no. of devices being already connected.
I have referred to the set of status codes that android documentation provides for gatt connection -> https://developer.android.com/reference/android/bluetooth/BluetoothGatt#summary but none corresponds to any such scenario.
I could see that there exists some set of hidden error codes that exists on BLE connection. For example i get status code 19 (Bluetooth LE: undocumented error code 19) which is not listed in the official documentation nor exists in BluetoothGatt. Another such example is of error code 133 (https://devzone.nordicsemi.com/f/nordic-q-a/45969/gatt-client-leaks-in-android-ble).

I could find the list of these error codes here but is there any documentation available around this? https://android.googlesource.com/platform/external/bluetooth/bluedroid/+/5738f83aeb59361a0a2eda2460113f6dc9194271/stack/include/gatt_api.h

The limit also varies for different devices so taking an assumption of 7 devices getting connected at max seems a bit inappropriate as this value could be lower in some while higher in different android devices.

Can anyone help me on the same providing me a way out to identify max connection limit reached on my device?

philips77, 2020-04-29:
I would need to make some test. I don't think any error is thrown in that case, perhaps just LogCat message is printed, at least I didn't see any by accident.

jainsameeksha, 2020-04-29:
> I would need to make some test. I don't think any error is thrown in that case, perhaps just LogCat message is printed, at least I didn't see any by accident.

Do u see any error code coming from Android for the same scenario?

jainsameeksha, 2020-05-23:
I agree, can u share that Logcat message what you are talking about?

philips77, 2022-03-21:
I'm closing the issue due to inactivity.

closed 2022-03-21 by philips77