nordicsemi/Android-BLE-Library #193: Sending notifications seems not allowing to send more after the first one

Atternatt, 2020-05-20:
Hello,

We have been trying the sendNotification call and after getting a success log we are not able to send a second notification. I'll copy the snipet of code that we are using to send the notification.

```
fun sendNotification(intent: Intent) {
      coroutineScope.launch {
          val deferred = async {
              logger.d(TAG, "sendNotification with intent: ${intent.toString()}")
              sendNotification(serverNotificationCharacteristic, intent.formattNotification())
                  .await()
              logger.d(TAG, "sendNotification DON'T FINISH")
          }
          deferred.await()
          logger.d(TAG, "sendNotification finished")
      }
  }

```

the returned logs are:
```
D/: Target initialized
D/: Data written to descr. 00002902-0000-1000-8000-00805f9b34fb, value: (0x) 01-00
D/: Notifications enabled
D/: Target initialized
D/: sendNotification with intent: Intent { act=com.mobilejazz.ble.ACTION_NOTIFICATION (has extras) }
D/: [Server] Sending notification to 46df2ad2-c827-47ae-9932-b64c3da77adc
D/: server.notifyCharacteristicChanged(device, 46df2ad2-c827-47ae-9932-b64c3da77adc, false)
D/: [Server callback] Notification sent (status=0)
D/: sendNotification with intent: Intent { act=com.mobilejazz.ble.ACTION_NOTIFICATION (has extras) }
D/: sendNotification with intent: Intent { act=com.mobilejazz.ble.ACTION_NOTIFICATION (has extras) }
```

as you can see 
```[Server callback] Notification sent (status=0)``` is shown (from the Ble library) but somehow 
```sendNotification finished```  nor ```sendNotification DON'T FINISH``` hasn't been shown and we are making a synchronous call so it should be reached.

We also tried to call enqueue() instead of await() but the behaviour is exactly the same. 

Our Bluetooth device is receiving the first notification and after that, we can not send more notifications.

It looks, following the logs, that we are enabling the notifications correctly for the correct characteristics so we are a little bit lost about that.

philips77, 2020-05-22:
Thank you for creating the issue. I'll look into it.

philips77, 2020-05-22:
Intersting. 
This line
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/51bfdc3754f4ecceb4f4dfcf21abed4b4f5a064a/ble/src/main/java/no/nordicsemi/android/ble/BleManagerHandler.java#L2572-L2577
is called, but 
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/51bfdc3754f4ecceb4f4dfcf21abed4b4f5a064a/ble/src/main/java/no/nordicsemi/android/ble/BleManagerHandler.java#L2599-L2609
seems not producing log, which means `request` is either `null` or has a different type. Could you put a breakpoint there and see?

philips77, 2020-05-22:
Besides, as I can see you're using the library in version 2.2, I'd like to ask for some feedback, if you have a minute:
1. What do you like / don't like about the lib and the API?
2. How easy was it to migrate (if you were using 2.1 before)?
3. Do you have any feature requests?

joselufo, 2020-05-22:
Hi @philips77, thanks for the detailed report. I'm Jose, a coworker of @Atternatt that we are working together on the same project.

We already try to debug it and we saw what you said. The `request` instance is `null` in the `private void notifyNotificationSent(@NonNull final BluetoothDevice device)` method that you described before. 

Our main concern is, is an internal issue from the library? or is something that we did not configure properly?

Regarding the feedback:
1. What do you like/don't like about the lib and the API?

The library has a really nice API to work with the Android Bluetooth stack that everybody knows that it's not the best thing if you compare with other platforms, for example, iOS. 

I don't like:
* The documentation, because there is none.
* Also, I would like to have read-only access to the Android Bluetooth callbacks. 

2. How easy was it to migrate (if you were using 2.1 before)?

We didn't migrate as we started from scratch in this project. 

3. Do you have any feature requests?

Honestly, documentation. It's pretty complex to understand how the library works inside without documentation. Most of the time you need to enter the library code to be able to figure out what you need to do.

philips77, 2020-05-22:
Thank you. It's true, that documentation could be better. The API should be documented in code, and there's an example on GitHub how to use it, but what's inside is a mystery. The assumption was that users should not care, a black box. But as you can see, sometimes a big needs to be traced down.

The `request` should not be null. Do you call `clearQueue()` at some point? If not, I need to look into it and find out where it gets nullified and why.

joselufo, 2020-05-22:
We are not calling to`clearQueue()` at any moment. If there is something more that we can help you, we will happy to do it.

joselufo, 2020-05-28:
@philips77 any update? we are in a hurry to have it fixed. Do you know a rough due date?

philips77, 2020-05-28:
Ok, I'm on it.

philips77, 2020-05-28:
Hello, 
I did a quick test by sending notifications like that:
```java
for (int i = 0; i < 5; i++) {
	sendNotification(testChar, new byte[]{0x00, 0x05})
		.done(device -> log(Log.INFO, "Notification sent"))
		.enqueue();
}
```
and I received the notification 5 times, which suggest an issue with how the lib handles the way you call `sendNotification`. Did you debug when the `request` gets nullified? Is it an issue with threading?

philips77, 2020-05-28:
I also tried with `await()` and got the same results. Are you doing something simultaneously to sending notifications?

philips77, 2020-06-24:
Hello, did you manage to solve the issue? Sorry for this very long delay.

joselufo, 2020-07-03:
@philips77 sorry for the late response. Yes, we fixed it. It was a bad use of the library. 

Nevertheless, it's nice if the library can handle this situation and at least throw an exception if the Queue is blocked. or not adding more operations.

philips77, 2020-07-03:
It's hard to detect a blocked queue. BLE callbacks should always be called, but in some situations they are not, and there's no obvious action to take at that point. I'm glad you solved it. I'll think how to improve the lib in that manner.

closed 2020-07-03 by philips77