nordicsemi/Android-BLE-Library #195: First 4 messages are fast, the rest is 1 sec delayed

benlarsendk, 2020-05-29:
Using this for an android application, I'm trying to debug why we get 0x08 errors at random times. To make sure we are not doing anything wrongfully on the app-side, we're using the example code from the readme. While doing this I set up some test code, the hardware is simply returning the received value, and when the app receives this, it sends it back to the board, so it's literally just playing ping-pong with a hello message. 

I notice that the first four messages are going fast, below 1s. But all consecutive messages are taking around 1s. I see the hardware is receiving and returning instantly, so my guess is that it's somewhere in the application, and after 40-140 messages the connection drops. 

The connection drop _might_ be an issue with our hardware, but the delay between messages seems to be the app. 

The BLEManager: https://pastebin.com/hebN6Cn4
The BLEService: https://pastebin.com/D0GbMiWz

Can anyone explain why we're having such a large delay with the messages?

philips77, 2020-06-24:
Hello,
Sorry for the very long delay. This looks like a change of connection interval. If you have logs, new connection parameters should be printed there. If it's not that, could you post the relevant part of the logs here?

benlarsendk, 2020-06-25:
Hi, 

Actually we finally solved this yesterday. We bought an NRF sniffer and saw we had a lot of requests to write from the Android device, and it took a long time for the board to respond. We solved this by changing the minimum connection interval to the lowest setting (15ms) and maximum to the highest (4s). This solved the problem, but we'd, of course, need to find the right settings somehow :) 

So the issue was _not_ with the library, but with the settings for RN4870 chip.

philips77, 2020-06-25:
Glad to hear that! I'm closing the issue than.

closed 2020-06-25 by philips77