nordicsemi/Android-BLE-Library #197: setNotificationCallback not getting called sometimes

casey55, 2020-06-05:
We are connecting to a peripheral that will display your temperature.  As soon as you push the button on the device it reports it's data and it seems like it is reporting it's data before we have had a change to set our callback.  Sometime the callback will get called though, but it seems pretty random.  Out initialize() looks like this.

```            
override fun initialize() {
    super.initialize()
    enableNotifications(characteristic).enqueue()
    setNotificationCallback(characteristic).with { _, data ->
        Log.i("ble", "data $data")
    }
}
````  

We are scanning for the device.  When we find it we connect to it and set everything up, but it seem too late most of the time.   I have tried setting the requestConnectionPriority to high, but does not appear to help.

Are there any other priorities to set that can possibly make the connection faster?

Thanks

philips77, 2020-06-24:
Hello,
Sorry for such long delay. Did you try changing the order of `enableNotifications` and `setNotificationCallback` invocations?
```kotlin
override fun initialize() {
    // super.initialize() // This is not required
    setNotificationCallback(characteristic).with { _, data ->
        Log.i("ble", "data $data")
    }
    enableNotifications(characteristic).enqueue()
}
```

philips77, 2020-06-30:
Assuming your issue has been fixed and closing. Good luck with the project!

closed 2020-06-30 by philips77