nordicsemi/Android-BLE-Library #200: beginAtomicRequestQueue not working as expected with a split() request.

andreasmakeable, 2020-06-12:
When adding a split-write request followed by a waitForNotification request, the wait request is started when the first packet is sent, even on multi-packet writes. This causes the write request to never send the remaining packets until the wait request times out.

philips77, 2020-06-12:
Hello,
Could you post code snippet and logs from LogCat?
Which library version are you using?

andreasmakeable, 2020-06-12:
Thanks for the quick reply!

I'm using version 2.2.0-beta01.

I pasted a snip of a function I use to send a command and await a response.
https://pastebin.com/RPb848PE

This is the LogCat output for the timber prints I inserted.
[image]

Hope you can help.

andreasmakeable, 2020-06-12:
Just tried the new stable 2.2.0 release, same thing.

philips77, 2020-06-12:
Could you also override the `log` method in your `BleManager` and log what you get there?

philips77, 2020-06-12:
Btw, there's `trigger` method in the `WaitForValueChangeRequest`:
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/51bfdc3754f4ecceb4f4dfcf21abed4b4f5a064a/ble/src/main/java/no/nordicsemi/android/ble/WaitForValueChangedRequest.java#L126-L130
Despite that the lib should not behave as it does in your case, you should use this method, as it ensures the notification callback is set before the request is actually sent. Sometimes the response is received before the lib transitions from one request to another and the `waitForNotification` will be processed after the notification was received. Could you check if it helps in your case?
Instead of using atomic queue, use
```kotlin
waitForNotification(tx)
  .before {
     Timber.d("Read: wait for notification is starting.")
  }
  .trigger(
    writeCharacteristic(rx!!, s.toByteArray())
      .before {
         Timber.d("Write: write characteristic is starting.")
      }
      .split { device: BluetoothDevice, data: ByteArray?, index: Int ->
         Timber.d("Write: split packet[$index] of ${data?.toString(Charsets.UTF_8)} was sent.")
      }
      .fail { device, status ->
         Timber.d("Write: Failed with status: $status")
         // [...]
      }
      .with { device, data ->
        // [...]
      }
  )
  .merge { output: DataStream, lastPacket: ByteArray?, index: Int ->
     // [...]
  }
  .timeout(5000L) // Timeout if 5 seconds between each packet happens.
  .fail { device, status ->
     Timber.d("Read: Failed with status: $status")
     // [...]
  }
  .with { _, data ->
     // [...]
  }
  .enqueue()
```

andreasmakeable, 2020-06-12:
I will have to get back to you monday, as I won't have the possibility to communicate with the peripheral until then.
Thanks a lot for the quick responses today.

philips77, 2020-06-12:
Sure. I'll try to reproduce your issue now with the use of atomic queue. This looks like a bug.

andreasmakeable, 2020-06-12:
I'll try your suggestion on monday too. Thanks

andreasmakeable, 2020-06-12:
Yeah, it seems like a bug. I checked by inserting breakpoints into the split implementation in the WriteRequest, and it gets split correctly, and correctly marks when the request is finished, but somehow the queue proceeds to the next request before the split-write is marked as complete. Hope it helps with debugging the issue.

andreasmakeable, 2020-06-15:
Hey,
I just tried with the trigger, and same problem. The operation on which it triggers is not fully sent when it is a split-write operation, before the waitForNotification happens.

Here is my logs as requested: https://pastebin.com/KEUiRrqd

Here is the revised BLE logic: https://pastebin.com/E6mCv3qm

andreasmakeable, 2020-06-17:
@philips77 any news on the issue?
Right now I am having to do a nasty workaround to get the logic working, as the split writes does not work in combination with either queues or triggers.

philips77, 2020-06-17:
It's next in my queue after I complete my current task. I'll try to look into this today evening. Sorry, that your have to wait :(

andreasmakeable, 2020-06-23:
@philips77 any updates?

philips77, 2020-06-23:
Sorry. I didn't have time to check it yet. I'll try d today/tomorrow.

philips77, 2020-06-24:
I found the root cause. I'll fix it now and will release a version. I'll notify you when the version is out, so you could give it a spin.

andreasmakeable, 2020-06-24:
Awesome. I'll try it when the release is out

philips77, 2020-06-24:
Could you check the *bugfix/200* branch? I'm sure it fixes the case when split write was set as a trigger for `waitForNotification`, but I'm not sure if the original issue is also fixed, with atomic queue. I'd like it to be verified before I release a new version.

To include the lib as a module, follow this: https://github.com/NordicSemiconductor/Android-BLE-Library#as-a-library-module

andreasmakeable, 2020-06-24:
I'll try it out

andreasmakeable, 2020-06-24:
@philips77 I can verify that it now works as expected with a split-write operation as a trigger for waitForNotification.
It does not work however with my previous (non-optimal) approach of sending a split-write in an atomicRequestQueue, and then awaiting notifications, here it still only sends the first split-package before proceeding to the next request in the queue.

philips77, 2020-06-25:
How about now? I find out that when the request enqueued to the main queue, not to the currently running request queue. With the above fix it should work in both your cases now.
And I'd like to thank you for your help. Much appreciated.

andreasmakeable, 2020-06-26:
@philips77 I'll try it out today

philips77, 2020-06-27:
I found one issue with atomic queue. A failed operation should fail the whole queue and following operations should not be started. I'll fix this on Monday.

andreasmakeable, 2020-06-30:
@philips77 thanks for the updates. I'll use the develop branch until your PR is merged into master. Cheers!

philips77, 2020-06-30:
Released as 2.2.1.

PR #206 (2020-06-29): Fixing chunked requests when awaiting request
This PR tries to fix #200. 

**Problem:**
When the manager was awaiting a notification or atomic queue was started, the library was unable to send requests split into multiple chunks. `operationInProgress` wasn't cleared, as `awaitingRequest` was not `nil`, causing following requests to be queued, but not executed. The atomic queue or awaiting requests would timeout eventually.

**Solution:**
The PR resets the `operationInProgress` each time a request is added to the beginning of the queue.
closed 2020-06-30 by philips77