nordicsemi/Android-BLE-Library #201: Connect doesn't work

trix0, 2020-06-12:
Hi, I have a wierd issue, I have spend some time trying to fix it but I am not sure whats the problem.

I have a window where I have displayed all discovered devices from Scanner. When I click on the device I would like to connect to it. I pass return from getdevice of ScanResult to myblemanager connect function and nothing happens.

I have tested code before using instrumented test. I had no problem connecting to the device.

The only difference is that in instrumented test I scan and as soon as I find the device I connect to it.

In the case above I store all the Scanned devices in Array and then connect to them based on my needs.

I would attach some logs but MybleManager does not produce any logs. I have no idea why. Maybe you can spot the mistake in my MyManager code. 

Thanks

This is the code of my blemanager. Sorry I had to obfuscate some parts.

```
public class MyManager extends BleManager   {
    String TAG="MyManager";

    public HotUUID HotUUIDs;

    public BluetoothGattCharacteristic
            first,
            second,
            third,
            fourth,
            fifth,
            sixth,
            seventh,
            LogService_Data,
            LogService_Racp;

    public MyHotManager(@NonNull final Context context,@NonNull final AppEnums.AppType Apptype) {
        super(context);
        HotUUIDs=new HotUUID();
        this.AppType=Apptype;
        DManager=new DManager(Apptype);
    }

    @NonNull
    @Override
    protected BleManagerGattCallback getGattCallback() {
        return new MyManagerGattCallback();
    }

    @Override
    public void log(final int priority, @NonNull final String message) {
        Log.d("BleManager", "[BLE LOG] " + message);
    }

    /**
     * BluetoothGatt callbacks object.
     */
    private class MyManagerGattCallback extends BleManagerGattCallback {

        // This method will be called when the device is connected and services are discovered.
        // You need to obtain references to the characteristics and descriptors that you will use.
        // Return true if all required services are found, false otherwise.
        @Override
        public boolean isRequiredServiceSupported(@NonNull final BluetoothGatt gatt) {
            final BluetoothGattService customService=gatt.getService(HotUUID.customService);
            final BluetoothGattService customServiceTwo=gatt.getService(HotUUID.customServiceTwo);
            final BluetoothGattService customServiceThree=gatt.getService(HotUUID.customServiceThree);
            final BluetoothGattService LogService=gatt.getService(HotUUID.LogService);

            if (customService ==null) {
                Log.e(TAG,"Patient Service not found");
                return false;
            }
            if (customServiceTwo ==null) {
                Log.e(TAG,"Diagnosis Service not found");
                return false;
            }
            if (customServiceThree ==null) {
                Log.e(TAG,"Oximeter Service not found");
                return false;
            }
            if (LogService ==null) {
                Log.e(TAG,"Log service not found");
                return false;
            }
            Log.i(TAG,"Found all the services");

            first=customService.getCharacteristic(HotUUIDs.firstUUID);
            second=customService.getCharacteristic(HotUUIDs.secondUUID);
            third=customService.getCharacteristic(HotUUIDs.thirdUUID);
            fourth=customService.getCharacteristic(HotUUIDs.fourth);

            seventh=customServiceThree.getCharacteristic(HotUUIDs.seventhUUID);

            fifth=customServiceTwo.getCharacteristic(HotUUIDs.fifthUUID);
            sixth=customServiceTwo.getCharacteristic(HotUUIDs.sixthUUID);

            LogService_Data=LogService.getCharacteristic(HotUUIDs.LogService_DataUUID);
            LogService_Racp=LogService.getCharacteristic(HotUUIDs.LogService_RacpUUID);

            if(first==null){
                Log.e(TAG,"first service not found");
                return false;
            }
            if(second==null){
                Log.e(TAG,"second service not found");
                return false;
            }
            if(third==null){
                Log.e(TAG,"third service not found");
                return false;
            }
            if(fourth==null){
                Log.e(TAG,"fourth service not found");
               // return false;
            }
            if(seventh==null){
                Log.e(TAG,"seventh service not found");
                return false;
            }
            if(fifth==null){
                Log.e(TAG,"fifth service not found");
                return false;
            }
            if(sixth==null){
                Log.e(TAG,"sixth service not found");
                return false;
            }
            if(LogService_Data==null){
                Log.e(TAG,"LogService_Data service not found");
                return false;
            }
            if(LogService_Racp==null){
                Log.e(TAG,"LogService_Racp service not found");
                return false;
            }

            return true;
        }

        // If you have any optional services, allocate them here. Return true only if
        // they are found.
        @Override
        protected boolean isOptionalServiceSupported(@NonNull final BluetoothGatt gatt) {
            return super.isOptionalServiceSupported(gatt);
        }

        // Initialize your device here. Often you need to enable notifications and set required
        // MTU or write some initial data. Do it here.
        @Override
        protected void initialize() {
            // You may enqueue multiple operations. A queue ensures that all operations are
            // performed one after another, but it is not required.
            beginAtomicRequestQueue()
                    .add(requestMtu(128) // Remember, GATT needs 3 bytes extra. This will allow packet size of 244 bytes.
                            .with((device, mtu) -> log(Log.INFO, "MTU set to " + mtu)))
                    .add(enableNotifications(seventh))

                    .add(enableNotifications(fifth))
                    .add(enableNotifications(sixth))
                    .add(enableIndications(sixth))

                    .add(enableNotifications(LogService_Data))
                    .add(enableNotifications(LogService_Racp))
                    .add(enableIndications(LogService_Racp))

                    .done(device -> {
                        log(Log.INFO, "Target initialized");
                    })

                    .enqueue();

            setNotificationCallback(seventh).with(new callback(){
                @Override
                public void Data(float s,float p) {

                }
            });

            setNotificationCallback(fifth).with(new RACPCallback(){
                @Override
                public void Data(sis result, Enums.Part Part) {
                    DdManager.AddD(result,Part);
                    DEvents.RecivedD(result,Part);

                }

            });

            setNotificationCallback(LogService_Data).with(new LogRACPCallback(){
                @Override
                public void LogData(LogMessage LogMessage) {
                    LogEvents.RecivedLog(LogMessage);
                }

            });

        }

        @Override
        protected void onDeviceDisconnected() {
            HotUUIDs=new HotUUID();
        }
    };

   
    public void abort() {
        cancelQueue();
    }

}
class HotUUID {
    final static UUID customService = UUID.fromString("00000001-FCD0-4CA4-B01B-5F1887095FEB");
    final static UUID firstUUID=UUID.fromString("00000002-FCD0-4CA4-B01B-5F1887095FEB");
    final static UUID secondUUID=UUID.fromString("00000003-FCD0-4CA4-B01B-5F1887095FEB");
    final static UUID thirdUUID=UUID.fromString("00000004-FCD0-4CA4-B01B-5F1887095FEB");
    final static UUID fourth=UUID.fromString("00000005-FCD0-4CA4-B01B-5F1887095FEB");

    final static UUID customServiceTwo = UUID.fromString("00000101-FCD0-4CA4-B01B-5F1887095FEB");
    final static UUID fifthUUID=UUID.fromString("00000102-FCD0-4CA4-B01B-5F1887095FEB");
    final static UUID sixthUUID=UUID.fromString("00001822-0000-1000-8000-00805f9b34fb");

    final static UUID customServiceThree = UUID.fromString("00001822-0000-1000-8000-00805f9b34fb");
    final static UUID seventhUUID   = UUID.fromString("00001822-0000-1000-8000-00805f9b34fb");

    final static UUID LogService   =          UUID.fromString("00000201-FCD0-4CA4-B01B-5F1887095FEB");
    final static UUID LogService_DataUUID   = UUID.fromString("00000202-FCD0-4CA4-B01B-5F1887095FEB");
    final static UUID LogService_RacpUUID  =  UUID.fromString("00002A52-0000-1000-8000-00805F9B34FB");
}
```

philips77, 2020-06-15:
Delayed connection should work, unless you restarted Bluetooth in between, or the devices you want to comment to are changing Bluetooth address and no longer listen to connect requests sent for this Mac address.

trix0, 2020-06-15:
Sorry, I actually don't know what was the problem. I came to work today and IT works. Thanks for your help @philips77

closed 2020-06-15 by trix0