nordicsemi/Android-BLE-Library #203: Connect to a device without doing a previous scan

c4software, 2020-06-14:
Hi,

I have a strange bug, my application is simple :

- Scan for a device
- Connect and write to a characteristic.
- Waiting for a notification.

Everything is working fine if I do a scan before try to connect… But if I save the device MAC address and try to reconnect directly to it after a reboot for example its not working.

Any idea why ?

My Manager connect : 
```kotlin
 manager.connect(getBleDevice(address))
            .timeout(10000)
            .retry(3, 100)
            .fail { device, status ->
                Log.d("BLE MANAGER", "${device.address} ($status)")
                notifyJacketState.close(Exception())
            }
            .enqueue()

```

My code to obtain « the device » from the Mac address is

```kotlin
    private fun getBleDevice(address: String): BluetoothDevice {
        val mBluetoothAdapter = BluetoothAdapter.getDefaultAdapter()
        return mBluetoothAdapter.getRemoteDevice(address)
    }
```

Nothings special in logs :

```logs
2020-06-14 22:35:48.933 15328-15479/com.sample.ble D/BleManager: [BLE LOG] Connecting...
2020-06-14 22:35:48.934 15328-15479/com.sample.ble D/BleManager: [BLE LOG] gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, LE 1M)
2020-06-14 22:35:48.934 15328-15479/com.sample.ble D/BluetoothGatt: connect() - device: D0:70:AA:FB:C8:8D, auto: false
2020-06-14 22:35:48.934 15328-15479/com.sample.ble D/BluetoothGatt: registerApp()
2020-06-14 22:35:48.934 15328-15479/com.sample.ble D/BluetoothGatt: registerApp() - UUID=6b4ba21b-b45f-46ec-b2f2-19e93d1af30a
2020-06-14 22:35:48.935 15328-16037/com.sample.ble D/BluetoothGatt: onClientRegistered() - status=0 clientIf=11
2020-06-14 22:35:48.956 15328-15328/com.sample.ble D/BLE OBSERVER: onDeviceConnecting D0:70:AA:FB:C8:8D
2020-06-14 22:35:58.932 15328-15328/com.sample.ble D/BleManager: [BLE LOG] Cancelling connection...
2020-06-14 22:35:58.932 15328-15328/com.sample.ble D/BleManager: [BLE LOG] gatt.disconnect()
2020-06-14 22:35:58.932 15328-15328/com.sample.ble D/BluetoothGatt: cancelOpen() - device: D0:70:AA:FB:C8:8D
2020-06-14 22:35:58.935 15328-15328/com.sample.ble D/BleManager: [BLE LOG] Disconnected
2020-06-14 22:35:58.935 15328-15328/com.sample.ble D/BLE MANAGER: D0:70:AA:FB:C8:8D (-5)
2020-06-14 22:35:58.936 15328-15328/com.sample.ble D/BLE OBSERVER: onDeviceDisconnected D0:70:AA:FB:C8:8D (0)
```

c4software, 2020-06-14:
I have tested my apk with different devices (Pixel 3, Pixel 4) and the behavior is always the same. 

The « direct » connection is working only if I do a scan before (in my app or via the system UI). 

If someone had an idea why...

philips77, 2020-06-15:
This is expected not to work without scanning, unless your device is bonded or it's using a public Bluetooth address. In Bluetooth the Mac address has actually 49 bits, out of which you can specify only 48 using Android API. The last bit is for address type: public or private. During scanning Android saves the last bit of each scanned advertising packet, so it knows which one to use on connection. When you restart phone or Bluetooth, this security database is lost, unless the device is bonded.

c4software, 2020-06-15:
Interesting, I was not aware of that! Looks like I will have to scan « each time » I want to interact with my device.

c4software, 2020-06-16:
After scanning I now don't have any issue.

closed 2020-06-16 by c4software