nordicsemi/Android-BLE-Library #204: How to implement a timeoutable waitForNotification?

kabiskac, 2020-06-16:
Hi,

If I understood it correctly, it's possible to enqueue requests like waitForNotification with a timeout using TimeoutableRequest, but since I am a beginner, it's not clear from the documentation how I should do it.

What I would like to do is to enqueue a writeCharacteristic, then enqueue a waitForNotification and if no answer comes from the periphery in x time, then some code should be executed.

Could someone please explain?

I hope I am writing at the right place.

philips77, 2020-06-24:
Hello,
Here's the code snipper:
```java
waitForCharacteristic(...)
   .trigger(writeCharacteristic(...))
   .timeout(SOME_TIME)
   .enqueue()
```

kabiskac, 2020-06-24:
Thank you very much, it's a life saver!

closed 2020-06-24 by philips77

kabiskac, 2020-06-24:
Sorry for another question here, but how/where do I handle the `Failcallback`?