nordicsemi/Android-BLE-Library #210: v2.2.1 breaks connection on some devices, BLE not working at all.

andreasmakeable, 2020-07-02:
Released as 2.2.1.

_Originally posted by @philips77 in https://github.com/NordicSemiconductor/Android-BLE-Library/issues/200#issuecomment-651738341_

The original issue merged in PR#209 breaks connection attempts on some devices, where devices will no longer be able to enqueue any requests to characteristics after connecting. Verified to work when downgrading to v2.2.0.

philips77, 2020-07-02:
Can you describe how to reproduce this? What requests do you enqueue? Please provide log.

andreasmakeable, 2020-07-02:
LogCat from unsuccesfull connect and subsequent write request **(v2.2.1)**:

```
2020-07-02 12:32:55.518 27471-27822/com.metricorr.app D/MasterLinkConnectionNRF: BLEMANAGER_LOG: Connecting...
2020-07-02 12:32:55.519 27471-27822/com.metricorr.app D/MasterLinkConnectionNRF: BLEMANAGER_LOG: gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, LE 1M)
2020-07-02 12:32:56.377 27471-27508/com.metricorr.app D/MasterLinkConnectionNRF: BLEMANAGER_LOG: [Callback] Connection state changed with status: 0 and new state: 2 (CONNECTED)
2020-07-02 12:32:56.377 27471-27508/com.metricorr.app D/MasterLinkConnectionNRF: BLEMANAGER_LOG: Connected to FB:CC:FD:FE:FD:70
2020-07-02 12:32:56.378 27471-27508/com.metricorr.app D/MasterLinkConnectionNRF: BLEMANAGER_LOG: wait(300)
2020-07-02 12:32:56.681 27471-27471/com.metricorr.app D/MasterLinkConnectionNRF: BLEMANAGER_LOG: Discovering services...
2020-07-02 12:32:56.681 27471-27471/com.metricorr.app D/MasterLinkConnectionNRF: BLEMANAGER_LOG: gatt.discoverServices()
2020-07-02 12:32:56.810 27471-27508/com.metricorr.app D/MasterLinkConnectionNRF: BLEMANAGER_LOG: Connection parameters updated (interval: 7.5ms, latency: 0, timeout: 5000ms)
2020-07-02 12:32:56.955 27471-27508/com.metricorr.app D/MasterLinkConnectionNRF: BLEMANAGER_LOG: Services discovered
2020-07-02 12:32:56.958 27471-27508/com.metricorr.app D/[Koin]: | create instance for [type:Single,name:'RX_UUID', primary_type:'android.os.ParcelUuid']
2020-07-02 12:32:56.959 27471-27508/com.metricorr.app D/MasterLinkConnectionNRF: BLEMANAGER_LOG: Primary service found
2020-07-02 12:32:56.962 27471-27508/com.metricorr.app D/MasterLinkConnectionNRF$MasterLinkGattCallback: RX: android.bluetooth.BluetoothGattCharacteristic@152e745, TX: android.bluetooth.BluetoothGattCharacteristic@721ae9a
2020-07-02 12:32:56.964 27471-27508/com.metricorr.app W/BluetoothGatt: Unhandled exception in callback
    java.lang.NullPointerException: Attempt to invoke interface method 'boolean java.util.Deque.add(java.lang.Object)' on a null object reference
        at no.nordicsemi.android.ble.BleManagerHandler.enqueue(BleManagerHandler.java:1193)
        at no.nordicsemi.android.ble.Request.enqueue(Request.java:1166)
        at dk.makeable.metricorr.domain.bluetooth.MasterLinkConnectionNRF$MasterLinkGattCallback.initialize(MasterLinkConnectionNRF.kt:93)
        at no.nordicsemi.android.ble.BleManagerHandler$3.onServicesDiscovered(BleManagerHandler.java:1888)
        at android.bluetooth.BluetoothGatt$1$5.run(BluetoothGatt.java:330)
        at android.bluetooth.BluetoothGatt.runOrQueueCallback(BluetoothGatt.java:789)
        at android.bluetooth.BluetoothGatt.-wrap0(Unknown Source:0)
        at android.bluetooth.BluetoothGatt$1.onSearchComplete(BluetoothGatt.java:326)
        at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:110)
        at android.os.Binder.execTransact(Binder.java:682)
2020-07-02 12:32:56.977 27471-27823/com.metricorr.app D/MasterLinkConnectionNRF$Commander: RX: android.bluetooth.BluetoothGattCharacteristic@152e745
    
    [          1593685976.979 27471:27823 E/MasterLinkConnectionNRF$MLApi$suspendCancellingIfNeeded$$inlined$suspendCancellableCorojava.lang.NullPointerException: Attempt to invoke interface method 'boolean java.util.Deque.add(java.lang.Object)' on a null object reference
        at no.nordicsemi.android.ble.BleManagerHandler.enqueue(BleManagerHandler.java:1193)
        at no.nordicsemi.android.ble.Request.enqueue(Request.java:1166)
        at dk.makeable.metricorr.domain.bluetooth.MasterLinkConnectionNRF$Commander.sendRaw(MasterLinkConnectionNRF.kt:365)
        at dk.makeable.metricorr.domain.bluetooth.MasterLinkConnectionNRF$MLApi$authenticate$2.invokeSuspend(MasterLinkConnectionNRF.kt:446)
        at dk.makeable.metricorr.domain.bluetooth.MasterLinkConnectionNRF$MLApi$authenticate$2.invoke(Unknown Source:10)
        at dk.makeable.metricorr.domain.bluetooth.MasterLinkConnectionNRF$MLApi$suspendCancellingIfNeeded$$inlined$suspendCancellableCoroutine$lambda$1.invokeSuspend(MasterLinkConnectionNRF.kt:677)
        at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
        at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:56)
        at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:571)
        at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:738)
        at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:678)
        at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:665)
2020-07-02 12:32:56.981 27471-27819/com.metricorr.app W/System.err:     at no.nordicsemi.android.ble.BleManagerHandler.enqueue(BleManagerHandler.java:1193)
2020-07-02 12:32:57.014 27471-27854/com.metricorr.app D/MasterLinkConnectionNRF: BLEMANAGER_LOG: Connection parameters updated (interval: 48.75ms, latency: 0, timeout: 5000ms)
```

LogCat from succesfull connect and subsequent write request **(v2.2.0)**:

```
2020-07-02 12:27:10.634 26786-26877/com.metricorr.app D/MasterLinkConnectionNRF: BLEMANAGER_LOG: Connecting...
2020-07-02 12:27:10.635 26786-26877/com.metricorr.app D/MasterLinkConnectionNRF: BLEMANAGER_LOG: gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, LE 1M)
[... cut ...]
2020-07-02 12:30:39.181 4754-27116/? D/BatteryStatsService: !@ mLastInfo WifiActivityEnergyInfo{ timestamp=607597980 mStackState=3 mControllerTxTimeMs=12030 mControllerTxTimePerLevelMs=[] mControllerRxTimeMs=508427 mControllerIdleTimeMs=6058287 mControllerEnergyUsed=152258371 }
2020-07-02 12:30:39.182 4754-27116/? D/BatteryStatsService: !@ latest WifiActivityEnergyInfo{ timestamp=607708215 mStackState=3 mControllerTxTimeMs=12033 mControllerTxTimePerLevelMs=[] mControllerRxTimeMs=509998 mControllerIdleTimeMs=6067271 mControllerEnergyUsed=152661189 }
2020-07-02 12:30:39.182 4754-27116/? D/BatteryStatsService: !@ delta timePeriodMs : 110235 txTimeMs : 3 rxTimeMs :1571 idleTimeMs : 8984
```

philips77, 2020-07-02:
Thank you very much.

andreasmakeable, 2020-07-02:
No problem @philips77 
I Will have to revert to version 2.2.0, and therefore the split-write functionality from PR#209 again, I will do a workaround until you get to the bottom of this connection problem.

Just for info:
I can consistently reproduce the issue on a Samsung Galaxy S8, S10, and on a OnePlus phone, Pixel devices seems to work fine with v2.2.1. Hope you can use that information for faster debugging.

I have a client waiting for their app, so any steps to fix this issue soon will be much appreciated!

philips77, 2020-07-02:
So I checked the logs and the code, and here're my findings.
When service discovery is complete, you reach line:
```
[...]
at no.nordicsemi.android.ble.BleManagerHandler$3.onServicesDiscovered(BleManagerHandler.java:1888)
[...]
```
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/e666df15f4a659242fd4f9629cec8b0aeeea8350/ble/src/main/java/no/nordicsemi/android/ble/BleManagerHandler.java#L1888
At this point `initQueue` is not *null*:
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/e666df15f4a659242fd4f9629cec8b0aeeea8350/ble/src/main/java/no/nordicsemi/android/ble/BleManagerHandler.java#L1852-L1853
and `initInProgress` is *true*:
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/e666df15f4a659242fd4f9629cec8b0aeeea8350/ble/src/main/java/no/nordicsemi/android/ble/BleManagerHandler.java#L1841

In the `initialize()` you enqueue a request, which calls:
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/e666df15f4a659242fd4f9629cec8b0aeeea8350/ble/src/main/java/no/nordicsemi/android/ble/BleManagerHandler.java#L1190-L1196
Looks, like `initInProgress` is in fact *true*, but the `initQueue` is *null*. 
This could happen if you called `cancelQueue()` or enqueued `refreshCache()` request.

philips77, 2020-07-02:
Could you put a breakpoint in line 1193 and in all places `initQueue` gets nullified? I wonder where and when it happens, and how does it work in 2.2.0.
Did you try cleaning your build? For me this looks like an issue with Android Studio or caching...

andreasmakeable, 2020-07-02:
I'll try it out, and update you with the results @philips77 
Thanks for the quick responses

andreasmakeable, 2020-07-02:
I found the source of the problem in:
[](https://github.com/NordicSemiconductor/Android-BLE-Library/blob/e666df15f4a659242fd4f9629cec8b0aeeea8350/ble/src/main/java/no/nordicsemi/android/ble/BleManagerHandler.java#L2805-L2808)

andreasmakeable, 2020-07-02:
I can't say why it only happens on v2.2.1, but It consistently fails on my Galaxy s10 development device, and consistently succeeds as expected in v2.2.0

philips77, 2020-07-02:
I think the issue is a bit different. In my app I request MTU change and enable notifications in `initialize()`, but the second request begins before the MTU is set. That means at some point the second request gets fired when it shouldn't. So I think I might have replicated this on Samsung S8. Hope this is the root cause of your issue as well.

philips77, 2020-07-02:
Ok, I have a candidate for change.
Could you check branch *bugfix/210*?

andreasmakeable, 2020-07-02:
@philips77 I'll check it out

andreasmakeable, 2020-07-02:
@philips77 I do requestMtu (+enqueue), and then in the next line enable notifications (+enqueue), so maybe the problem is the same as yours.

andreasmakeable, 2020-07-02:
@philips77 Your fix is not working, now multiple requests are not performed, probably because the lines your changed in bugfix/210 causes the next request to not be run.

I can see when debugging that isInitInProgress is set to true multiple times AFTER initially being false, I suspect this causes the new bugfix you released to drop requests because of line #3188 [here](https://github.com/NordicSemiconductor/Android-BLE-Library/commit/2267f8f633b2fed5d959efc8172aef4dc9121f84#diff-1cb4b2f422b908df1d18fc1354011fc1L3188)

philips77, 2020-07-02:
Ok, how about now?

philips77, 2020-07-02:
I created a PR. We can discuss your following findings there.

andreasmakeable, 2020-07-03:
I'll try it out @philips77

PR #211 (2020-07-03): Executing init queue fixed
This PR fixes #210 - an issue with 2.2.1, where init queue fails on phones running older Android versions.

On Android 4.3-4 5 and 8-9 the lib tries to enable indications for Service Changed characteristic as a first task in initialization queue. The request is aborted when the devices are not bonded, as not relevant. But the abort was causing next request to be started immediately. This accumulated problems, where the `initQueue` could be nullified before any next request was enqueued, causing NPE. The proposed fix disables executing init queue until initialization  [... cut]
closed 2020-07-03 by philips77