nordicsemi/Android-BLE-Library #216: Null Pointer Exception for v2.2.0

SarthakWhoop, 2020-07-07:
Android app seems to be crashing in the background when trying to reconnect back to the BLE device. It crashes especially when trying to enable notifications for characteristics. I have this in the code base
`(enableNotifications(btGattChr) as SimpleRequest).await()` 

This is the stack trace when the exception happens:
[image]

I am also seeing an uptick in a similar NPE on our users when the app sends a disconnect command. Below is the stack trace:
[image]

Any help/suggestions appreciated!

philips77, 2020-07-07:
Hi,
Please try 2.2.3, which is the current version now. I'm on vacations at three moment, but I'll look into it when I get back to work.
Also, why do you cast result returned from `enableNotifications` to `SimpleRequest`?

prfarlow1, 2020-07-08:
Developer from the same project here. I'd like to add that 98% of the time this crash occurs, the app is in the background.

SarthakWhoop, 2020-07-08:
We have updated to v2.2.3 and will report back if we start to see a similar NPE in background crashes on our end. In the mean time, I'll also remove the cast to `SimpleRequest`

philips77, 2020-07-10:
Any success?

SarthakWhoop, 2020-07-13:
Nope. We upgraded to v2.2.3 and I still notice crashes with the same above exceptions. 
```
Fatal Exception: java.lang.NullPointerException: Attempt to invoke virtual method 'void l.a.a.a.k2.t(android.bluetooth.BluetoothDevice, int)' on a null object reference
       at no.nordicsemi.android.ble.BleManagerHandler.nextRequest(BleManagerHandler.java:3214)
       at no.nordicsemi.android.ble.BleManagerHandler.enqueue(BleManagerHandler.java:1225)
       at no.nordicsemi.android.ble.Request.enqueue(Request.java:1179)
       .....
       at android.os.Handler.handleCallback(Handler.java:873)
       at android.os.Handler.dispatchMessage(Handler.java:99)
       at android.os.Looper.loop(Looper.java:214)
       at android.os.HandlerThread.run(HandlerThread.java:65)
```

```
Fatal Exception: java.lang.NullPointerException: Attempt to invoke virtual method 'void android.bluetooth.BluetoothGatt.disconnect()' on a null object reference
       at no.nordicsemi.android.ble.BleManagerHandler.internalDisconnect(BleManagerHandler.java:599)
       at no.nordicsemi.android.ble.BleManagerHandler.nextRequest(BleManagerHandler.java:2916)
       at no.nordicsemi.android.ble.BleManagerHandler.enqueue(BleManagerHandler.java:1195)
       at no.nordicsemi.android.ble.Request.enqueue(Request.java:1166)
       at no.nordicsemi.android.ble.TimeoutableRequest.enqueue(TimeoutableRequest.java:76)
       .....
       at android.os.Handler.handleCallback(Handler.java:883)
       at android.os.Handler.dispatchMessage(Handler.java:100)
       at android.os.Looper.loop(Looper.java:214)
       at android.os.HandlerThread.run(HandlerThread.java:67)
```

SarthakWhoop, 2020-07-13:
I also happen to stumble upon a similar issue created by another user here: https://github.com/NordicSemiconductor/Android-BLE-Library/issues/107

philips77, 2020-07-15:
Copy that. I'll look into that when on back from vacation.

SarthakWhoop, 2020-08-10:
Are there any updates on this?

philips77, 2020-08-11:
Could you try on develop branch? The fix should help with crashes, but I'm not sure how could you get to the state that it was an issue. Serious race conditions.

SarthakWhoop, 2020-08-12:
Ok. Thank you for fixing this. 
I am going to update the library and report back next week or so if I notice crashes happening for users.

philips77, 2021-08-31:
I'm closing the issue due to inactivity. Hope it works for you now!

closed 2021-08-31 by philips77