nordicsemi/Android-BLE-Library #217: BLE peripheral failing to connect to the central phone running Android OS 9 (Pie)

SarthakWhoop, 2020-07-13:
The app issues a `connect(bluetoothDevice)` command as shown below to the Nordic BLE library v2.2.3 after a user manually clicks "Connect" from the UI in the app. 

```Kotlin
connect(bluetoothDevice).useAutoConnect(true)
                    .usePreferredPhy(PhyRequest.PHY_LE_2M_MASK)
                    .before { log.debug("connect request starting execution id $id") }
                    .fail { _, status -> log.debug("connect request id $id failed ${status.toRequestFailureMessage()}") }
                    .enqueue()
```
I also make a check before using LE 2M PHY if the device is compatible. If the device is incompatible, I call like shown below:
```Kotlin
connect(bluetoothDevice).useAutoConnect(true)
                    .before { log.debug("connect request starting execution id $id") }
                    .fail { _, status -> log.debug("connect request id $id failed ${status.toRequestFailureMessage()}") }
                    .enqueue()
```
I get a `ConnectionObserver` callback `onDeviceFailedToConnect` in my custom class which implements the interface with a reason `-1` 

```Java
/** The reason of disconnection is unknown. */
int REASON_UNKNOWN = -1;
```
This is however only affecting Android OS 9 and most probably 9.1 users. We have noticed that if you upgrade the OS to the latest stable OS 10, the issue seems to go away. We also have noticed that for devices running OS 9, if you pair first through Bluetooth OS settings and then try again through within the app, you can successfully pair to the BLE peripheral. 

For your information, we upgraded Nordic library from v2.1.1 to 2.2.0 and the issue was still existing. We thought upgrading to v2.2.3 might resolve but in vain. Is there something that we are doing wrong?

philips77, 2020-07-15:
Could you post some logs from logcat? Make sure you log what you get in `log` method in the `BleManager`.

SarthakWhoop, 2020-07-15:
Here you go @philips77: 
The logs below are when the BLE peripheral is fast advertising (20 msec) without whitelisting
```
2020-07-15 14:01:08.406 18773-20678/com.whoop.android.dev.qa V/BleManager(183085907): (thread Whoop Strap Handler) Connecting...
2020-07-15 14:01:08.407 18773-20678/com.whoop.android.dev.qa D/BleManager(183085907): (thread Whoop Strap Handler) gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, LE 2M)
2020-07-15 14:01:13.426 18773-18989/com.whoop.android.dev.qa D/BleManager(183085907): (thread Binder:18773_4) [Callback] Connection state changed with status: 133 and new state: 0 (DISCONNECTED)
2020-07-15 14:01:13.431 18773-18989/com.whoop.android.dev.qa W/BleManager(183085907): (thread Binder:18773_4) Error: (0x85): GATT ERROR
2020-07-15 14:01:13.432 18773-18989/com.whoop.android.dev.qa W/BleManager(183085907): (thread Binder:18773_4) Connection attempt timed out
2020-07-15 14:01:13.437 18773-18989/com.whoop.android.dev.qa D/BleManager(183085907): (thread Binder:18773_4) gatt.close()
```

Below are the logs when I pair first through Bluetooth settings and then click connect from the custom Android app implementing `BleManager` which is a successful connection attempt
```
2020-07-15 14:08:53.624 22550-23082/com.whoop.android.dev.qa V/BleManager(154205239): (thread Whoop Strap Handler) Connecting...
2020-07-15 14:08:53.625 22550-23082/com.whoop.android.dev.qa D/BleManager(154205239): (thread Whoop Strap Handler) gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, LE 2M)
2020-07-15 14:08:53.632 22550-22563/com.whoop.android.dev.qa D/BleManager(154205239): (thread Binder:22550_2) [Callback] Connection state changed with status: 0 and new state: 2 (CONNECTED)
2020-07-15 14:08:53.632 22550-22563/com.whoop.android.dev.qa I/BleManager(154205239): (thread Binder:22550_2) Connected to E0:9E:19:D3:61:D6
2020-07-15 14:08:53.634 22550-22563/com.whoop.android.dev.qa D/BleManager(154205239): (thread Binder:22550_2) wait(1600)
2020-07-15 14:08:55.241 22550-22550/com.whoop.android.dev.qa V/BleManager(154205239): Discovering services...
2020-07-15 14:08:55.242 22550-22550/com.whoop.android.dev.qa D/BleManager(154205239): gatt.discoverServices()
2020-07-15 14:08:55.251 22550-22563/com.whoop.android.dev.qa I/BleManager(154205239): (thread Binder:22550_2) Services discovered
2020-07-15 14:08:55.260 22550-22563/com.whoop.android.dev.qa V/BleManager(154205239): (thread Binder:22550_2) Primary service found
2020-07-15 14:08:55.269 22550-22563/com.whoop.android.dev.qa I/BleManager(154205239): (thread Binder:22550_2) Service Changed characteristic found on a bonded device
2020-07-15 14:08:55.271 22550-22563/com.whoop.android.dev.qa D/BleManager(154205239): (thread Binder:22550_2) gatt.setCharacteristicNotification(00002a05-0000-1000-8000-00805f9b34fb, true)
2020-07-15 14:08:55.275 22550-22563/com.whoop.android.dev.qa V/BleManager(154205239): (thread Binder:22550_2) Enabling indications for 00002a05-0000-1000-8000-00805f9b34fb
2020-07-15 14:08:55.276 22550-22563/com.whoop.android.dev.qa D/BleManager(154205239): (thread Binder:22550_2) gatt.writeDescriptor(00002902-0000-1000-8000-00805f9b34fb, value=0x02-00)
2020-07-15 14:08:55.434 22550-22563/com.whoop.android.dev.qa I/BleManager(154205239): (thread Binder:22550_2) Data written to descr. 00002902-0000-1000-8000-00805f9b34fb, value: (0x) 02-00
2020-07-15 14:08:55.435 22550-22563/com.whoop.android.dev.qa I/BleManager(154205239): (thread Binder:22550_2) Service Changed notifications enabled
2020-07-15 14:08:55.438 22550-22563/com.whoop.android.dev.qa V/BleManager(154205239): (thread Binder:22550_2) Requesting new MTU...
2020-07-15 14:08:55.439 22550-22563/com.whoop.android.dev.qa D/BleManager(154205239): (thread Binder:22550_2) gatt.requestMtu(247)
2020-07-15 14:08:55.589 22550-22563/com.whoop.android.dev.qa I/BleManager(154205239): (thread Binder:22550_2) MTU changed to: 247
2020-07-15 14:08:55.592 22550-22550/com.whoop.android.dev.qa I/WhoopStrapGatt: onReady called by BleManager
2020-07-15 14:08:55.602 22550-23082/com.whoop.android.dev.qa D/BleManager(154205239): (thread Whoop Strap Handler) gatt.setCharacteristicNotification(61080004-8d6d-82b8-614a-1c8cb0f8dcc6, true)
2020-07-15 14:08:55.607 22550-23082/com.whoop.android.dev.qa V/BleManager(154205239): (thread Whoop Strap Handler) Enabling notifications for 61080004-8d6d-82b8-614a-1c8cb0f8dcc6
2020-07-15 14:08:55.607 22550-23082/com.whoop.android.dev.qa D/BleManager(154205239): (thread Whoop Strap Handler) gatt.writeDescriptor(00002902-0000-1000-8000-00805f9b34fb, value=0x01-00)
2020-07-15 14:08:55.883 22550-22563/com.whoop.android.dev.qa I/BleManager(154205239): (thread Binder:22550_2) Data written to descr. 00002902-0000-1000-8000-00805f9b34fb, value: (0x) 01-00
2020-07-15 14:08:55.884 22550-22563/com.whoop.android.dev.qa I/BleManager(154205239): (thread Binder:22550_2) Notifications enabled
2020-07-15 14:08:55.888 22550-22563/com.whoop.android.dev.qa D/BleManager(154205239): (thread Binder:22550_2) gatt.setCharacteristicNotification(61080003-8d6d-82b8-614a-1c8cb0f8dcc6, true)
2020-07-15 14:08:55.891 22550-22563/com.whoop.android.dev.qa V/BleManager(154205239): (thread Binder:22550_2) Enabling notifications for 61080003-8d6d-82b8-614a-1c8cb0f8dcc6
2020-07-15 14:08:55.891 22550-22563/com.whoop.android.dev.qa D/BleManager(154205239): (thread Binder:22550_2) gatt.writeDescriptor(00002902-0000-1000-8000-00805f9b34fb, value=0x01-00)
[... cut ...]
2020-07-15 14:09:08.782 22550-23082/com.whoop.android.dev.qa V/BleManager(154205239): (thread Whoop Strap Handler) Writing characteristic 61080002-8d6d-82b8-614a-1c8cb0f8dcc6 (WRITE COMMAND)
2020-07-15 14:09:08.782 22550-23082/com.whoop.android.dev.qa D/BleManager(154205239): (thread Whoop Strap Handler) gatt.writeCharacteristic(61080002-8d6d-82b8-614a-1c8cb0f8dcc6)
2020-07-15 14:09:08.786 22550-22663/com.whoop.android.dev.qa I/BleManager(154205239): (thread Binder:22550_4) Data written to 61080002-8d6d-82b8-614a-1c8cb0f8dcc6, value: (0x) AA-10-00-57-23-0D-17-01-A2-CE-01-00-2B-00-00-00-0F-F3-C2-C5
```

SarthakWhoop, 2020-07-17:
Hi @philips77! Any updates on this issue? Seems to be affecting a lot of users on Android OS 9 and 8 too.

philips77, 2020-07-17:
Hi, i'm on vacations until 27th June.
Few questions:
1. Does it happen 100% times?
2. Can you connect using nRF Connect?
3. Did you try taking sniffer logs?
4. Can you connect with other peripherals, e.g. our samples from the SDK?
This error 133 looks like some issue with the phone or your fw. This library can't fix this issue, as it's the system that connects and fails.

SarthakWhoop, 2020-07-17:
Thank you @philips77 for your response! To answer your questions:

1. It happens 100% times when using the library. Like I mentioned, to resolve this, I have to first pair the BLE peripheral from the Bluetooth settings and then open the custom app which implements the Nordic Library and then click "Connect". I have posted the relevant logs above in the thread.
2. I am sometimes able to and sometimes not able to connect using the nRF Connect app. It is sort of flaky for me. Also, feel free to correct me but doesn't nRF Connect app **NOT** use the `Android-BLE-Library`?
3. I haven't tried that yet. But I can do that and post them here. 
4. Did not try this. I'll take a look at the samples from the SDK and post the results back here. 

To your comment, once you update from Android 9 to Android 10, the issue is no longer valid. I can successfully connect and bond to the peripheral on Android OS 10 with the same code!

philips77, 2020-07-17:
3. No, nRF Connect was implemented long before this lib was created. The lib was created based on the experience I had gained on it.

If it works for you in the same phone on Android 10, i guess Samsung has fixed some bugs...

SarthakWhoop, 2020-07-27:
I was going through issues and found a similar one [here](https://github.com/NordicSemiconductor/Android-BLE-Library/issues/159). Do you think calling from any other thread except the UI thread might be the cause here too for OS 9? I tried to wrap my call for `connect(bluetoothDevice)` in the MainLooper but I got the same GATT 133 error

It looks like the main call for `connectGatt` is happening on the UI thread but the callbacks are happening on a Binder thread. 

```
2020-07-27 09:58:04.301 16317-16317/com.w.a.d.q  D/BleManager(90922442): gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, LE 2M)
2020-07-27 09:58:09.330 16317-16684/com.w.a.d.q  D/BleManager(90922442): (thread Binder:16317_8) [Callback] Connection state changed with status: 133 and new state: 0 (DISCONNECTED)
2020-07-27 09:58:09.331 16317-16684/com.w.a.d.q W/BleManager(90922442): (thread Binder:16317_8) Error: (0x85): GATT ERROR
2020-07-27 09:58:09.331 16317-16684/com.w.a.d.q  W/BleManager(90922442): (thread Binder:16317_8) Connection attempt timed out
2020-07-27 09:58:09.335 16317-16684/com.w.a.d.q  D/BleManager(90922442): (thread Binder:16317_8) gatt.close()
```

philips77, 2020-07-27:
Yes, all callbacks are executed in a binder. In the `connectGatt` is it now possible to specify the callback thread, but that only means the binder will call the callback on this looper. This just adds a delay, which causes troubles when you write and get notifications from the same characteristic (data is reused for sending and receiving and may be overwritten).

SarthakWhoop, 2020-08-07:
@philips77 Can you link me to the SDK samples for me to try with?

philips77, 2020-08-07:
All our Sdks are here: http://developer.nordicsemi.com/

SarthakWhoop, 2020-08-10:
I am trying to capture sniffer logs from nRF 51 dongle by following the guide [here](https://infocenter.nordicsemi.com/index.jsp?topic=%2Fug_sniffer_ble%2FUG%2Fsniffer_ble%2Finstalling_sniffer_plugin.html) and I am currently stuck at Step 4. I successfully completed step 3 and below is the result when I run the following command `sh nrf_sniffer_ble.sh --extcap-interfaces` 
```
usage: dirname path
extcap {version=3.0.0}{display=nRF Sniffer for Bluetooth LE}{help=https://www.nordicsemi.com/Software-and-Tools/Development-Tools/nRF-Sniffer-for-Bluetooth-LE}
interface {value=/dev/cu.usbmodem0006804533651}{display=nRF Sniffer for Bluetooth LE}
control {number=0}{type=selector}{display=Device}{tooltip=Device list}
control {number=1}{type=string}{display=Passkey / OOB key}{tooltip=6 digit temporary key or 16 byte Out-of-band (OOB) key in hexadecimal starting with '0x', big endian format. If the entered key is shorter than 16 bytes, it will be zero-padded in front'}{validation=\b^(([0-9]{6})|(0x[0-9a-fA-F]{1,32}))$\b}
control {number=2}{type=string}{display=Adv Hop}{default=37,38,39}{tooltip=Advertising channel hop sequence. Change the order in which the siffer switches advertising channels. Valid channels are 37, 38 and 39 separated by comma.}{validation=^\s*((37|38|39)\s*,\s*){0,2}(37|38|39){1}\s*$}{required=true}
control {number=3}{type=button}{role=help}{display=Help}{tooltip=Access user guide (launches browser)}
control {number=4}{type=button}{role=restore}{display=Defaults}{tooltip=Resets the user interface and clears the log file}
control {number=5}{type=button}{role=logger}{display=Log}{tooltip=Log per interface}
value {control=0}{value= }{display=All advertising devices}{default=true}
```
I installed the latest Wireshark version for Mac and there was no `Interface Toolbars` option under View. I found this post [here](https://devzone.nordicsemi.com/f/nordic-q-a/38641/wireshark-is-not-listing-the-interface-toolbars)  which says that the guide was written for older version so I downgraded my wireshark to v2.4.2 and also tried on v2.4.6 but in vain. 

I tried rebooting the Wireshark software and also the laptop but the option did not appear. I also unplugged and plugged back the dongle to see any changes in the list of interfaces (after clicking Refresh) but it did not list `nRF Sniffer for Bluetooth LE`

Could you please tell me how do I capture the sniffer logs? Is there an updated guide for that?
This issue is really picking steam and we have confirmed on our end, the custom BLE peripheral successfully connects 100% of the time without using the BLE library but does not connect when the library is used.

philips77, 2020-08-11:
You may also find this [post](https://devzone.nordicsemi.com/f/nordic-q-a/32128/sniffer-on-osx---revisited) useful, but it's regarding older version. We now support only Python 3.6+.

philips77, 2020-08-11:
For more, please ask your question on DevZone. Out Support Team will be more capable helping you.

philips77, 2020-08-11:
Btw, in the initial post you wrote you are trying to connect with preferred PHY LE 2M:
```java
connect(bluetoothDevice).useAutoConnect(true)
                    .usePreferredPhy(PhyRequest.PHY_LE_2M_MASK)
                    .before { log.debug("connect request starting execution id $id") }
                    .fail { _, status -> log.debug("connect request id $id failed ${status.toRequestFailureMessage()}") }
                    .enqueue()
```
Did you try removing this line and using the default PHY?

SarthakWhoop, 2020-08-11:
Ok. I'll make a ticket on DevZone to get support on sniffer logs. 

I have also tried removing the line `.usePreferredPhy(PhyRequest.PHY_LE_2M_MASK)` and it **doesn't** work.
Also just to be clear, I first check if the central (phone) supports PHY LE 2M with this call `bluetoothAdapter.isLe2MPhySupported();` and if it returns `true` then only I set the PreferredPhy otherwise I don't

SarthakWhoop, 2020-08-19:
I am having some trouble getting sniffer logs on my mac and currently in touch with a support engineer on the DevZone [here](https://devzone.nordicsemi.com/f/nordic-q-a/64832/nrf-sniffer-option-not-available-in-wireshark-for-mac/264811#264811). In the mean time has there been any developments on this? I am curious to know if you were also able to reproduce or investigate any thing on your end?

SarthakWhoop, 2020-08-26:
I have some sniffer logs available for you to debug. 

1. `OS9_pairing_failure.pcapng` : The BLE peripheral was fast advertising (20 msec) without whitelisting and the custom app was running on Samsung S10 (OS 9)
2. `OS9_pairing_success_via_bluetooth_settings.pcapng`: I first paired the BLE Peripheral from Bluetooth settings and then clicked connect from the custom app running on S10 which was a success! 

[SnifferLogs.zip](https://github.com/NordicSemiconductor/Android-BLE-Library/files/5131473/SnifferLogs.zip)

SarthakWhoop, 2020-08-31:
Hi @philips77!
Is there any status update on this? Would like to get this resolved ASAP if possible.

philips77, 2020-09-04:
Hi, did you try adding `.retry(times, delay)` to connect request?

SarthakWhoop, 2020-09-15:
Hi @philips77!
We tested the recommendation of adding `.retry(times, delay)` but it did not help!

SarthakWhoop, 2020-09-17:
@philips77 is there anything else that I can try or any updates on this issue?

closed 2022-03-21 by philips77

ahmed560, 2023-02-07:
Hello Aleksander, 
I have faced an issue in a BLE protocol. 
The issue in detail:
-       SoC: nRF52810

-       Softdevice: S112 v6.1.0

We are using nrf52 as a peripheral device on one side and the nrf connect application as a central device on the other. While the Application discovers devices for the first time, peripherals occur in the scan list, and then the central initializes the connection status and the bond is created.

After many connect statuses between the same peers, while we are trying to connect the GATT, we get GATT error 133. After I click on connect, it takes 30 secs of delay and after that, we receive Error 133. so when we remove data from the scanner and we established again the bond the connection will be fixed but the problem appears one more time after many other connect statuses. 

thank you for any idea which can help me to find the solution to fix this issue.