nordicsemi/Android-BLE-Library #221: Unable to disconnect when ConnectRequest is processing.

shiehnpin, 2020-07-26:
Hi there,

I'm facing the behaviour that I'm not sure whether is the limitation or the issue.  Firstly, I invoke a connect request to the device which isn't there at the time. Then I call disconnect() after the connect request is enqueued. But no matter how the disconnect request is invoked by await() or enqueue(), the process is postponed until the connect request timeout. 

My question is: Is it not allowed to disconnect while the connecting is processing?

Mobile: Pixel 3a
Library version: 2.2.3

philips77, 2020-07-27:
Currently there is no abort feature in any of the requests. But this seems to be a valid use case. I'll add this to the backlog.

philips77, 2020-07-27:
Task: Add `abort()` method to `ConnectRequest`.

shiehnpin, 2020-07-28:
In my opinion, aborting the connecting process in some stages might cause issues based on my previous experience. And most of them are coming with error code 133. I think to allow cancellation before handshaking initiated during the connecting process should be enough. What do you suggest?

Ted-kim-dev, 2020-10-15:
I have the same problem.

closed 2022-03-21 by philips77