nordicsemi/Android-BLE-Library #223: Question: How to avoid registering to notifications for optimisation?

Christophe668, 2020-08-05:
Hi,

I discovered that on the nRF Connect app when we connect to a device, previously activated notifications are already activated without having to execute the request. We can also see on the firmware that no write requests are done.

We would like to replicate that behavior to reduce the number of writes at the connection. 
My question is; how can we achieve such a thing? I looked deeply (probably not enough though) but I don't see anything, do you execute a read of the descriptor before setting it?

Would be amazing to have your help on this!

Thanks in advance

philips77, 2020-08-05:
Hello,
The nRF Connect app is enabling notifications [locally](https://developer.android.com/reference/android/bluetooth/BluetoothGatt#setCharacteristicNotification(android.bluetooth.BluetoothGattCharacteristic,%20boolean)) for all characteristics with NOTIFY/INDICATE properties. By enabling them locally, the client (`BluetoothGatt`) will receive any incoming notification. If a notification or an indication is received, the app assumes that they must have been enabled by another app/in a previous session, so it marks the CCCD as enabled and behaves as such.
In Bluetooth spec it is said, that server should remember the CCCD state in bonded devices, as far as I remember. I don't know, however, if it's not allows to do so without bonding. But with no bonding none of the sides can assume that the connected device is the same that enabled CCCD before, so by default they should be disabled. nRF Connect is a bit special in a sense here, as it's made for debugging.

Answering your question, you may call [`setCharacteristicNotification`](https://developer.android.com/reference/android/bluetooth/BluetoothGatt#setCharacteristicNotification(android.bluetooth.BluetoothGattCharacteristic,%20boolean)) in `isRequiredServiceSupported`, where you get the `BluetoothGatt` object and set notification callback in `initialize()`. This should work. Of course, assuming your device will be sending notifications without explicit CCCD write.

Christophe668, 2020-08-05:
Very clear, thanks a lot for taking the time to reply!

closed 2020-08-05 by philips77