nordicsemi/Android-BLE-Library #224: Fatal Exception: java.lang.IllegalArgumentException: Parameter specified as non-null is null: method kotlin.jvm.internal.Intrinsics.b, parameter device

dustedrob, 2020-08-10:
I've been seeing this crash in my app with Samsung devices. It seems that the BluetoothDevice object is coming up as null:

```
Fatal Exception: java.lang.IllegalArgumentException: Parameter specified as non-null is null: method kotlin.jvm.internal.Intrinsics.b, parameter device
       at com.myapp$MyBleManager$ZcGattCallback$initialize$1.onRequestCompleted(MyBleManager.java:2)
       at no.nordicsemi.android.ble.Request.lambda$notifySuccess$1(Request.java:1190)
       at no.nordicsemi.android.ble.Request.lambda$6KM5eyWvy4PFbDqmAqoy3fIWu5o(Request.java)
       at no.nordicsemi.android.ble.-$$Lambda$Request$6KM5eyWvy4PFbDqmAqoy3fIWu5o.run(-.java:4)
       at android.os.Handler.handleCallback(Handler.java:883)
       at android.os.Handler.dispatchMessage(Handler.java:100)
       at android.os.Looper.loop(Looper.java:237)
       at android.app.ActivityThread.main(ActivityThread.java:7857)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:493)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1076)

```
Is there anything that can be done to mitigate it? I'm currently on version **2.2.0-beta03**

philips77, 2020-08-11:
The current version is 2.2.3. Did you try this one?

philips77, 2020-08-11:
There was this fa0a697ed7d644b699cba7ec7b5791df98ffa4f2 commit some time ago, which should help with your issue.

roberto-betancourt-zwift, 2020-08-11:
Great I'll check it out. Thanks

dustedrob, 2020-10-28:
@philips77 We moved to 2.2.3 after you suggested it and we've been monitoring the crash rate since then. It went down considerably but now it's peaking again with mostly Android 11 devices. We have not made any changes to our app.

philips77, 2020-10-29:
I'm on leave until mid January. I'll look into it when I'm back. Do you have logs? Are they different?

dustedrob, 2020-11-03:
Here is the stacktrace from the.crash, it is still the same crash, I
believe:

Fatal Exception: java.lang.IllegalArgumentException: Parameter specified as
> non-null is null: method kotlin.jvm.internal.Intrinsics.b, parameter
> <anonymous parameter 0>
>        at
> com.app.android.ble.DefaultBleManager$MyGattCallback$initialize$1.onRequestCompleted(BlePeripheralBridge.java:2)
>        at
> no.nordicsemi.android.ble.Request.lambda$notifySuccess$1(Request.java:1203)
>        at
> no.nordicsemi.android.ble.Request.lambda$6KM5eyWvy4PFbDqmAqoy3fIWu5o(Request.java)
>        at
> no.nordicsemi.android.ble.-$$Lambda$Request$6KM5eyWvy4PFbDqmAqoy3fIWu5o.run(-.java:4)
>        at android.os.Handler.handleCallback(Handler.java:888)
>        at android.os.Handler.dispatchMessage(Handler.java:100)
>        at android.os.Looper.loop(Looper.java:213)
>        at android.app.ActivityThread.main(ActivityThread.java:8178)
>        at java.lang.reflect.Method.invoke(Method.java)
>        at
> com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:513)
>        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1101)

valeriyo, 2021-03-07:
I'm getting the same crash with (latest) version 2.2.4 -- "Parameter specified as non-null is null" in the `notifyFailed` callback:

```
no.nordicsemi.android.ble.Request.lambda$notifyFail$2 (Request.java:1218)
no.nordicsemi.android.ble.Request.lambda$notifyFail$2$Request (Request.java:2)
no.nordicsemi.android.ble.-$$Lambda$Request$MU5LE_ePYPVe9rzIdkv_E9MTNdc.run (-.java:2)
```

Happened on Samsung Galaxy Note20 5G and Galaxy S8 running Android 9.

valeriyo, 2021-04-10:
Hello? How about changing:
```
void onRequestFailed(@NonNull final BluetoothDevice device, final int status)
```
to 
```
void onRequestFailed(@Nullable final BluetoothDevice device, final int status)
```
?

philips77, 2021-04-12:
Hi,
Sorry for the very long delay. I'm currently involved in another project, but should be able to come back to the BLE library in next few weeks and look at all reported issues. Regarding your suggestion, I'd rather check why and where it gets nullified instead of allowing null. Your solution would break API.

valeriyo, 2021-05-25:
@philips77 wonder if you had any time to work on this issue? It's one of the larger contributors of crashes in our app.

philips77, 2021-05-26:
I'll be releasing new version of the library today (hopefully).

mbt-anhtuan-m, 2021-11-24:
Hi, the bug has not fixed yet ?
It will occur on Android 11 when user refuse to bond.

mecoFarid, 2022-01-25:
Is there any way to catch those exceptions without `Thread.UncaughtExceptionHandler`? Those exceptions are being nightmare. Simply because `notifySuccess` and other similar methods use `Handler.postDelayed(Runnable())` it is impossible wrap them with `try/catch`. Is there any workaround?

philips77, 2022-03-21:
Hello,
We've made number of releases since 2.2.x. Please, try the lates beta version. Also #371 adds `try/catch` blocks around user callbacks, but I think this issue should no longer happen.

closed 2022-03-21 by philips77