nordicsemi/Android-BLE-Library #226: How to connect to a BLE peripheral that may not be nearby at app startup?

usikder, 2020-08-14:
I'm building a BLE app where I want to connect to my peripheral whenever it comes in range. As I understand, autoconnect = true is intended for this use case. I'm having trouble doing this when the peripheral isn't nearby when my app starts up.

Since NordicBLE uses autoconnect=false for the first connection (https://github.com/NordicSemiconductor/Android-BLE-Library/blob/9ef7fcc9658e50c61b30ca592884ed558d61ddbf/ble/src/main/java/no/nordicsemi/android/ble/BleManager.java#L411) and only queues another connect call if the device was previously connected (https://github.com/NordicSemiconductor/Android-BLE-Library/blob/1b836ad846ef86fed35fc93807ff8e1056afd57f/ble/src/main/java/no/nordicsemi/android/ble/BleManagerHandler.java#L1812), if the peripheral isn't around on app startup, the connection request times out w/o enqueuing an autoconnect = true call.

Is this expected behavior? The only workaround I can think of is repeatedly making connection requests, which seems very inefficient.

philips77, 2020-08-15:
Hello,
First of all, in order to connect, you have to first scan for the device, or have it bonded. By creating `BluetoothDevice` without scanning, Android will assume public type of address if it didn't scan that device since Bluetooth was turned on and such connection request would fail even with this device nearby.
So my recommendation for you would be to start scanning, on Android Oreo using `PendingIntent` and on older versions with low power setting, and when the device is found, you initiate connection.

usikder, 2020-08-15:
Thanks for the response. This is with a device I'm already bonded to. To elaborate, on my app's startup, I enqueue a ConnectRequest to the bonded device with autoconnect = true. If the bonded device is nearby at the time, I get a successful connection - https://gist.github.com/usikder/7ee5e5c491f33bb212bc8fa440875741. If the bonded device isn't nearby at the time, the connection times out (https://gist.github.com/usikder/580facd809b510c3048662d0ac6961e5).

In the second case, I want my app to connect to the bonded peripheral whenever it does come nearby. AIUI, this requires a autoconnect = true connectGatt call, which Nordic's library doesn't seem to support today.

philips77, 2020-08-16:
Ok, I think you have a good point. If the device is bonded, autoConnect could be used also on the first connection, or rather when the first connection times out. As I wrote before, connection with this parameter set to false is much faster, therefore preferred.

usikder, 2020-08-16:
Agreed, enqueuing an autoConnect=true call if the initial autoConnect=false call fails would be nice.

closed 2022-03-21 by philips77