nordicsemi/Android-BLE-Library #235: Coroutine support

wcauchois, 2020-08-27:
Are there any plans to integrate [Kotlin coroutine](https://kotlinlang.org/docs/reference/coroutines-overview.html) support into this library? That way for most methods that return Requests, they could be executed as coroutines instead (possibly throwing an exception).

I currently have implemented this myself with code like the following:

```kotlin
    suspend fun writeCharacteristicSuspend(characteristic: BluetoothGattCharacteristic?, value: ByteArray) {
        characteristic?.writeType = BluetoothGattCharacteristic.WRITE_TYPE_NO_RESPONSE
        val result = CompletableDeferred<Unit>()
        writeCharacteristic(characteristic, Data(value))
                .with { _, data -> result.complete(Unit) }
                .invalid { result.completeExceptionally(RequestInvalidException) }
                .fail { _, status -> result.completeExceptionally(RequestFailedException("write characteristic", status)) }
                .enqueue()
        result.await()
    }
```

but it would be great to get first-class support in this library!

philips77, 2020-08-28:
We would like to have it Kotlin, yes. There was even a PR, where someone has converted it to Kotlin, but I think using automatic migration. The only problem is with finding resources.
Also, I am not yet fluent in Kotlin (I'm implementing a sample app for this lib in Kotlin to learn that language), so it may be difficult for me to keep the quality at the beginning. Kotlin likes doing things in 1 line, I would write it in 10 (Java needs 20, I guess).

philips77, 2020-08-28:
I bet version 3 will be in Kotlin.

nrobi144, 2021-01-27:
@philips77 I think with a `ble-coroutines` module it'd be possible even now. We've done a coroutine layer over the current implementation for scanning, reading and writing in one of our projects and could help out with this until kotlin becomes a first-class citizen 😀

DavidJRobertson, 2021-04-14:
Even this snippet alone is very useful (though of course you could go much further than this):
```kotlin
import android.bluetooth.BluetoothDevice
import no.nordicsemi.android.ble.Request
import no.nordicsemi.android.ble.callback.FailCallback
import no.nordicsemi.android.ble.callback.InvalidRequestCallback
import no.nordicsemi.android.ble.callback.SuccessCallback
import no.nordicsemi.android.ble.exception.BluetoothDisabledException
import no.nordicsemi.android.ble.exception.DeviceDisconnectedException
import no.nordicsemi.android.ble.exception.InvalidRequestException
import no.nordicsemi.android.ble.exception.RequestFailedException
import kotlin.coroutines.Continuation
import kotlin.coroutines.resume
import kotlin.coroutines.resumeWithException
import kotlin.coroutines.suspendCoroutine

private class SuspendingRequestCallback(
    val request: Request, val continuation: Continuation<Unit>
) : SuccessCallback, FailCallback, InvalidRequestCallback {

    override fun onRequestCompleted(device: BluetoothDevice) = continuation.resume(Unit)

    override fun onRequestFailed(device: BluetoothDevice, status: Int) {
        val exception = when (status) {
            FailCallback.REASON_DEVICE_DISCONNECTED -> DeviceDisconnectedException()
            FailCallback.REASON_BLUETOOTH_DISABLED -> BluetoothDisabledException()
            else -> RequestFailedException(request, status)
        }
        continuation.resumeWithException(exception)
    }

    override fun onInvalidRequest() = continuation.resumeWithException(InvalidRequestException(request))
}

suspend fun Request.runSuspending() {
    suspendCoroutine<Unit> { cont ->
        val callback = SuspendingRequestCallback(this, cont)
        this.done(callback).fail(callback).invalid(callback).enqueue()
    }
}
```

mecoFarid, 2022-01-25:
Just to point out the pitfall: 

There is an extension method in the library with the following pattern. Just to be clear any exception thrown from this method **can not** be caught because any code block provided to `done`, `invalid` etc. will be executed in a handler as `Runnable` and any exception thrown from `Runnable` **can not** be caught, so it just crashes the application. Do you guys have any specific reason to choose this pattern?

```
private suspend fun Request.suspendCancellable(): Unit = suspendCancellableCoroutine { continuation -> this
	.done { continuation.resume(Unit) }
	.invalid { continuation.resumeWithException(InvalidRequestException(this)) }
	.fail { _, status ->
		val exception = when (status) {
			FailCallback.REASON_BLUETOOTH_DISABLED -> BluetoothDisabledException()
			FailCallback.REASON_DEVICE_DISCONNECTED -> DeviceDisconnectedException()
			else -> RequestFailedException(this, status)
		}
		continuation.resumeWithException(exception)
	}
	.enqueue()
}
```

philips77, 2022-01-26:
What would you recommend? Are you saying, that 
```kotlin
continuation.resume(Unit)
```
may crash the app if the resumed call throws? Or where can this exception originates?

philips77, 2022-01-27:
We will try to replicate what you said.

What I don't understand is 
>  Just to be clear any exception thrown from this method can not be caught because any code block provided to done, invalid etc. 

The `continuation.resumeWithException(InvalidRequestException(this))` doesn't throw the exception. It notifies the suspended thread about the exception, but this can be called from any thread.

philips77, 2022-01-27:
To catch an exception returned this way you need to install `CoroutineExceptionHandler` like described e.g. here: https://medium.com/the-kotlin-chronicle/coroutine-exceptions-3378f51a7d33:
```kotlin
val errorHandler = CoroutineExceptionHandler { context, error ->
    when (error) {
        is SomeSillyExeption -> ...
        ...
   }
}
val scope = CoroutineScope(errorHandler)scope.launch {
    throw SomeSillyException()
}
// Will cause the 'errorHandler' to be invoked and our app won't crash.
```

philips77, 2022-01-27:
OK, we found the root cause. We will try to come up with a solution next week. Thank you for reporting the issue.

mecoFarid, 2022-01-30:
> The `continuation.resumeWithException(InvalidRequestException(this))` doesn't throw the exception. It notifies the suspended thread about the exception, but this can be called from any thread.

You are totally right about this but if the supervisor scope doesn't have any error hander (`CoroutineExceptionHandler`) this exception will crash the app. This is what I was trying to say and I think you already got what I meant

closed 2022-03-21 by philips77