nordicsemi/Android-BLE-Library #236: PHY not supported

pritam2102, 2020-08-28:
Got issue in only nokia 6.1 "PHY updated failed with status 6" how to resolve please help me.

philips77, 2020-08-28:
Could you share your logs and describe what you are doing exactly? What is the version if Android on that phone? Can you also post a screenshot from nRF Connect -> Device Information screen taken on this phone?

pritam2102, 2020-08-28:
Ble is connected but not getting a response from ble on mobile

Log:-   
Requesting preferred PHYs...
gatt.setPreferredPhy(LE 2M, LE 2M, coding option = No preferred)
PHY updated failed with status 6
Requested PHY not supported: 6
Connection parameters updated (interval: 45.0ms, latency: 0, timeout: 5000ms)

Device Info :- 
Device Name:- Nokia 6.1 
Version :- Android 10

[image]

philips77, 2020-08-28:
Well.. looks like one of the devices: your phone or the peripheral does not support PHY LE 2M. It's not an issue, I would say.

okbean, 2020-12-16:
see the constant GATT_REQUEST_NOT_SUPPORTED in class BluetoothGatt, the value is 0x6.
it means that，setPreferredPhy request is not support on nokia 6.1.

closed 2020-12-17 by philips77

habbasi-rsc, 2021-03-02:
@philips77 Should we check the Android version manually before using such this `.add(setPreferredPhy(PhyRequest.PHY_LE_2M_MASK, PhyRequest.PHY_LE_2M_MASK, PhyRequest.PHY_OPTION_NO_PREFERRED)`
?
Or the library does this internally?

Drjacky, 2021-03-02:
@philips77 Should we check the Android version manually before using such this .add(setPreferredPhy(PhyRequest.PHY_LE_2M_MASK, PhyRequest.PHY_LE_2M_MASK, PhyRequest.PHY_OPTION_NO_PREFERRED)
?
Or the library does this internally?

philips77, 2021-03-02:
The library should check it and also check if requested phy is supported.