nordicsemi/Android-BLE-Library #240: Should an auto connect re-perform initialisation?

huntc, 2020-09-10:
When connecting to a device with `useAutoConnect` set to `true`, I notice that on a link having been lost and then re-established, the initialisation sequence of discovering services and the `initialisation` function is performed. Is this necessary?

philips77, 2020-09-10:
Bonded devices should remember handle values (e.g. CCCD), but initialization does not necessarily mean only enabling notifications. It may be authentication, or anything. Besides, the device does not have to be bonded, or doesn't need to remember those values.
I think it doesn't harm to initialize every time + you also need to enable notifications locally, which is done together with sending 0x0X-00 to a remote CCCD.

huntc, 2020-09-10:
Thanks. Given that an auto reconnect implies a bond (you need a stable MAC address in practical terms), is it normal for BLE devices to reinitialise on reconnect then? Seems like a lot of chatter over the air.

philips77, 2022-03-21:
Closing this issue due to inactivity.

closed 2022-03-21 by philips77